1technophile/OpenMQTTGateway #958: esp32dev-rtl_433: Cannot make use of unparsed signals (unknown protocols)

TheChatty, 2021-05-24:
**Describe the bug**
Unparsed signals always differ entirely on each occurence thus are not yet useful.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash **esp32dev-rtl_433** flavor of [releases](https://github.com/1technophile/OpenMQTTGateway/releases).
2. [Connect CC1101](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib#wiring).
3. Setup WiFi + MQTT broker, [finetune CC1101](https://docs.openmqttgateway.com/use/rf.html#rtl-433-device-decoders).
4. Push button of a remote (which uses a yet unknown rtl_433 protocol).
5. Notice entirely different raw message each time.

**Expected behavior**
Receive a steady message to make use of (no matter how huge it is for unknown protocols).

**Environment:**
* OpenMQTTGateway V0.9.6

**Additional context**
So OpenMQTTGateway works basically. I see raw messages coming into the broker as well as at the serial connection.

The remotes I want to use now are [this](https://github.com/RFD-FHEM/RFFHEM/issues/906) ([commit](https://github.com/RFD-FHEM/RFFHEM/commit/981ae0e830a6d9fcc6b75d1312d0e5d41fbc7173)) and [this](https://github.com/RFD-FHEM/RFFHEM/issues/909) ([commit](https://github.com/RFD-FHEM/RFFHEM/commit/0f8ee748ca92c0f6bb7490444a92e6ef543968ad)). As you can see, both protocols have already been decoded in [another project](https://github.com/RFD-FHEM/SIGNALDuino) which is able to send data with a now known protocol.

It would be marvelous if OpenMQTTGateway, rtl_433 and SignalDuino would work together. Until then I'd be happy if the signals from my remotes would be decoded in a way leading to stable data and to be able to repeat those signals.

NorthernMan54, 2021-05-25:
@TheChatty The rtl_433 package is typically used with sensor devices, for devices like switches did you try the RCSwitch flavour of openMQTTGateway ?  If you compile the 'esp32dev-rf' environment it may support your switch.  And then you can also transmit as well

https://docs.openmqttgateway.com/use/rf.html#rcswitch-based-gateway

The 'Unparsed Signal' signal message is just indicating that the signal that was received is not supported by the rtl_433_ESP library, and that you may want to try one of the other receiver packages.

PS If you use the development branch of OpenMQTTGateway I have added the ability to switch between the various receiver modules available.  Details are here

https://github.com/1technophile/OpenMQTTGateway/pull/936

TheChatty, 2021-05-25:
Yesterday I was using latest released binary. The devices I'd like to control are a light/venti-combo and motorized projector canvas.

I did as you said:

- compiled 18c8524ae08bd9160702dd91880defa232cf3eab with 'default_envs = esp32dev-rf'
- published `-t .../commands/MQTTto433 -m {"mhz":433.920}` and got<br/>`N: Switching to RF Receiver`<br/>`N: Received json : {"Status":"Error"}` and no RF messages
- digging into f69906861dae2d44865ffef9536714257df99f19 I found you meant me to compile with `default_envs = esp32dev-multi_receiver`
- by reading the "[docs](f69906861dae2d44865ffef9536714257df99f19)" I found:
  - `mosquitto_pub -t ../commands/MQTTtoPilight/config -m {"active":"true","mhz":433.81}` lead to<br/>`Switching to Pilight Receiver: 433.81`
  - `mosquitto_pub -t ../commands/MQTTto433/config -m {"active":"true","mhz":433.82}` lead to<br/>`Switching to RF Receiver: 433.82`
  - `mosquitto_pub -t ../commands/MQTTtoRTL_433/config -m {"active":"true","mhz":433.83}` lead to<br/>`Switching to RTL_433 Receiver: 433.83`
  - `mosquitto_pub -t ../commands/MQTTtoRF2/config -m {"active":"true","mhz":433.84}` lead to<br/>` Switching to RF2 Receiver: 433.84`
- at the same frequency only RTL_433 displays receivings for both remotes (but as said above: always different for same key)
- RF displays decoded receivings for BeSmart remote (Yay!)
- `{"active":"false"}` also changes the receiver (I don't know if this is worth changing)
- there is too much name confusion (RF/RF2/RTL_433/433)

To sum it up: OpenMQTTGateway is not usable for my remotes. SignalDuino is capable of decoding hundreds of protocols (even with esp8266 or arduino plus CC1101) with an all-in-one firmware. Again, it would be cool if it could communicate with MQTT.

NorthernMan54, 2021-05-25:
I had assumed that the RF Receiver decoder would be able to handle all your switches as it is very robust for basic switches and is able to handle all of mine.  I have a collection of Hampton bay and other fans, and a fireplace that are all able to be handled by RF Receiver module.

The multi receiver feature is meant to be used when you want to receive signals with one of the decoders ( RF,RF2, PiLight or RTL_433 ) and transmit with another.  I use it to receive temperature sensor signals with the rtl_433 module and transmit with the RF Gateway.  You can directly transmit with RF, RF2 or PiLight ( RTL_433 doesn't have a transmit function ) while receiving signals with another module.

TheChatty, 2021-06-07:
I didn't know that RF is the most versatile decoder. I just chose it to test all four of them.

closed 2022-04-16 by stale[bot]