1technophile/OpenMQTTGateway #959: TSL2561 report ftcd as values

pauldogg, 2021-05-24:
Foot candles (ftcd) are reported as a string of text in home assistant, could it be reported as a value please?
[image]

And also could sensor names adopt the gateway name to help find sensors in HA?
Right now the TSL2561 has a sensor that appears as sensor.lux which is pretty generic, could it become sensor.omg01_lux or something like that?
Thanks!

1technophile, 2021-06-09:
> Right now the TSL2561 has a sensor that appears as sensor.lux which is pretty generic, could it become sensor.omg01_lux or something like that?

Maybe something with a unique id of the gateway?

> Foot candles (ftcd) are reported as a string of text in home assistant, could it be reported as a value please?

Could you point me how you updated your board, if you used Arduino IDE or PIO could you try to add "illuminance" [here](https://github.com/1technophile/OpenMQTTGateway/blob/2737a4cbf7c7202f50e8c81fbb5e80b3a2032cb3/main/ZmqttDiscovery.ino#L440) like the line above and see if it is better?

pauldogg, 2021-06-25:
Yeah with a unique id of the gateway would be ideal!
I used Arduino IDE to flash the board.
As far as the code, happy to try but i'm not really good at fiddling with code, could you be more explicit on what you'd like me to try please?

Thanks

closed 2022-04-16 by stale[bot]