1technophile/OpenMQTTGateway #980: LYWSD03MMC relevant data from connection get replaced by broadcast data

fhb, 2021-06-09:
**Describe the bug**
Although I set the esp32 to connect with every scan the temperature/humidity/battery data on mosquitto (version 1.5.7) is getting replaced by the data broadcasted by the LYWSD03MMC sensor on a scan: 
*Broadcast data:*
```
{"id":"A4:C1:38:17:5F:77","name":"LYWSD03MMC","rssi":-40,"servicedata":"30585b055a775f1738c1a408"}
```
*Relevant data:*
```
{"model":"LYWSD03MMC","id":"A4:C1:38:17:5F:77","tempc":26.35,"tempf":79.43,"hum":56,"volt":2.919,"batt":81.9}
```

Therefore the relevant data is not available to openhab or mqttthing on homebridge most of the time and errors are thrown. Also "onlysensors" and "whitelist" seem not to work properly (set through mqtt commands).

**To Reproduce**
Steps to reproduce the behavior:

- Hardware ESP32-WROOM-32, 3x LYWSD03MMC (with 1.0.0._0130 fw)
- Download OpenmQTTGateway Sources and load esp32dev_ble into the the arduino libraries
- Correct known bug on line 523 (missing function void....)
- #define OpenHABDiscovery true
- Install OpenMQTTGateway 0.9.6 on ESP32 dev with activated ZGateway_BLE through ArduinoIDE.
- Configure MQTT and WIFI
- set: scan to connect ratio to 1 ``mosquitto_pub  -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"scanbcnct":1}'``
- whitelist Sensors ``mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"white-list":[your MAC adresses]}'``
- trigger scan and hopefully connect too with ``mosquito_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"interval":0}'``
- also tried with ``mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"onlysensors":true}'``

**Expected behavior**
I would expect OpenMQTTGateway to only plublish the relevant data set after a successful connection and otherwise not push an mqtt message.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* Hardware ESP32-WROOM-32, 3x LYWSD03MMC (with 1.0.0._0130 fw)
* OpenMQTTGateway V0.9.6 with esp32dev_ble, most recent ArduinoIDE, MQTT-Broker Mosquitto V 1.5.7, 

* Possible workaround, earlier discussion:
Only solution would be a broadcasting firmware right now. Similar issues seem to have been discussed here: https://community.openmqttgateway.com/t/xiaomi-mijia-lywsd03mmc-esp32-support/775/49

1technophile, 2021-06-09:
Would it be possible to share what you get on the serial monitor after the scan?

> Only solution would be a broadcasting firmware right now. Similar issues seem to have been discussed here:

This solution is better in terms of battery consumption and quite easy to implement (less than 5min to flash the ATC firmware)

fhb, 2021-06-10:
Sure. It might be the connection failing from time to time, but I don't understand the serial output fully:

[omg_serialoutput.txt](https://github.com/1technophile/OpenMQTTGateway/files/6627802/omg_serialoutput.txt)

NorthernMan54, 2021-06-10:
I have a similar symptom with the DT24 Bluetooth voltmeter I added, and I found that two different message's were being sent on the same device topic.  I found that I was seeing a message from the BLE scan and a second different message from the device BT connect.

But I found that it did not cause an issue with my client as it was using home assistant mqtt discovery and it was ignoring the messages with missing fields.  Does openHab have something similar to the home assistant value template and the is_defined macro ?  In my setup that caused the messages to be ignored by the client.

```
{"stat_t":"+/+/BTtoMQTT/200507161361",
"name":"DT24-volt",
"uniq_id":"200507161361-DT24-volt",
"dev_cla":"power",
"val_tpl":"{{ value_json.volt | is_defined }}",
"unit_of_meas":"V",
"device":{"connections":[["mac","200507161361"]],"identifiers":["200507161361"],"manufacturer":"ATorch","model":"DT24-BLE","name":"DT24","via_device":"cc1101-dddddd"}}
```

h2zero, 2021-06-13:
@fhb I just added a commit to #979 that should address this issue, could test it and let me know the results?

1technophile, 2021-07-10:
@fhb you can use the macro `BLE_FILTER_CONNECTABLE`, by setting it to 1 you will not receive broadcasted messages from connectable devices.

closed 2021-07-10 by 1technophile