1technophile/OpenMQTTGateway #985: Issues OTA flashing Sonfoff RF Bridge

dkneisz, 2021-06-13:
Using the latest development branch I'm having issues to flash a Sonoff RF Bridge OTA. I get the following error:
```
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = espota
Uploading .pio\build\rfbridge-RF-remote\firmware.bin
08:58:14 [DEBUG]: Options: {'esp_ip': '192.168.0.25', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 26167, 'auth': '*****', 'image': '.pio\\build\\rfbridge-RF-remote\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True}
08:58:14 [INFO]: Starting on 0.0.0.0:26167
08:58:14 [INFO]: Upload size: 475184
08:58:14 [INFO]: Sending invitation to: 192.168.0.25
Authenticating...FAIL
08:58:14 [ERROR]: No Answer to our Authentication
*** [upload] Error 1
```
In the serial monitor I get:
```
Error[1]: E: Begin Failed
```
This is not a general problem with OTA since flashing an ESP32 is working fine. It also was working with Sonoff RF Bridge some weeks ago. I noticed that the compiled firmware.bin seems to be too large to flash OTA:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  39.8% (used 32564 bytes from 81920 bytes)
Flash: [======    ]  61.8% (used 471032 bytes from 761840 bytes)
```
So I went back in the GIT history until I found a commit where it was working. I finally landed on commit https://github.com/1technophile/OpenMQTTGateway/commit/31efcb32eeea9a2d4e9ddae5cae69ad0424ab7d6. With this one I was able to flash my Sonoff RF Bridge with OTA again. 
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  39.1% (used 31996 bytes from 81920 bytes)
Flash: [=====     ]  47.3% (used 360396 bytes from 761840 bytes)
```

It looks like commit https://github.com/1technophile/OpenMQTTGateway/commit/b8b99c6d5dcce37b6b2865a6fe4148f1c9a2d8d5 is causing the firmware.bin to become too large. I cherry picked all commits following this one and with them everything is working fine. So the issue seems to be coming from this commit only.

h2zero, 2021-06-13:
Thanks for bringing this up, it's strange that commit caused such a large increase in flash size.

1technophile, 2021-07-07:
The change of size is due to the code commented below:
[image]

We may just use a macro MQTTsetMQTT as proposed in the PR and let the other part of the code always here (MQTTsetWifi).

1technophile, 2021-07-21:
Done already

closed 2021-07-21 by 1technophile