1technophile/OpenMQTTGateway #994: WifiManager password is lost if we trigger a configuration reset of the gateway

1technophile, 2021-06-20:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
The WifiManager_password  is not required after a gateway reset by long press

**To Reproduce**
Steps to reproduce the behavior:
- Switch on the gateway on an ESP32
- Connect to the Wifi AP with a smartphone
- Enter WifiManager_password
- Configure it through the WM web portal
- Save the configuration
- The gateway connects to the wifi and mqtt broker
- Do a long press on TRIGGER_GPIO
- Remove the wifi AP from your smartphone
- Connect to the Wifi AP with a smartphone
- WifiManager_password is not needed anymore, it should be requested

**Expected behavior**
The WifiManager_password  should still be requested

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.6)

h2zero, 2021-06-21:
If I'm interpreting the steps correctly, the password would stored in the phones cache. So unless it changes on resetting I think this is normal behavior.

1technophile, 2021-06-21:
I was thinking about this at the beginning, but unfortunately, if I remove the network/erase it from my smartphone wifi list, I can still reconnect to it without the wifi password.
A good indicator could be the lock symbol close to the wifi symbol. My home wifi has it always, and the wifi of OMG disappears after an init by a long button press.

PR #1009 (2021-07-06): Erase stored settings in NVS when triggered.
## Description:
This resolves #994 where the esp32 would still save the client information after a factory reset is performed.

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit c5564696 (2021-07-06): Erase stored settings in NVS when triggered. (#1009)

This resolves #994 where the esp32 would still save the client information after a factory reset is performed.
closed 2021-07-06 by 1technophile