1technophile/OpenMQTTGateway #1005: Thingsboard as controller support

garudaonekh, 2021-07-01:
**Is your feature request related to a problem? Please describe.**
Thingsboard is cool and have a lot of features for developers to customize. I am using Thingsboard cloud

**Describe the solution you'd like**
The ability to push the collected data to Thingsboard thru MQTT publishing

**Describe alternatives you've considered**
Using Thingsboard gateway as a bridge is an option but this will require an additional Raspberry PI as Thingsboard gateway required RPI or Linux machine.

**Additional context**
If you don't have time for this, can you point me to the right direction. Or a better solution that I may miss

1technophile, 2021-07-01:
It would be interesting to add thingsboard support. I can help you do this.
I suppose that if you don't want to go through the gateway we need to leverage this [API](https://thingsboard.io/docs/reference/mqtt-api/) ?

garudaonekh, 2021-07-02:
> It would be interesting to add thingsboard support. I can help you do this.
> I suppose that if you don't want to go through the gateway we need to leverage this [API](https://thingsboard.io/docs/reference/mqtt-api/) ?

Correct. Their MQTT syntax are simple. The main thing is the DEVICE_TOKEN, which currently generated from the Thingsboard Cloud Server but it may have some self discovery method or something which I am not aware of.

garudaonekh, 2021-07-02:
In my current setup is

BLE Sensors->Your gateway->My Local MQTT Broker->Collected by Thingsboard Gateway

The above setup is fine for system with many sensors as Thingsboard Gateway has local backup for unsent data which is a good feature.

BUT I think for smaller setup, it's more efficient to send from your gateway to Thingsboard directly.

closed 2022-04-27 by stale[bot]