1technophile/OpenMQTTGateway #1033: LYWSD03MMC not connected when whitelist is enabled

MountComb, 2021-07-22:
**Describe the bug**

If an undiscovered LYWSD03MMC is added to BLE white list, it is not connected and can not read temperature data.

**To Reproduce**

1. add the LYWSD03MMC sensor to white list using retained MQTT message
2. restart the gateway
3. wait for a few scan / connect cycles
4. the sensor is discovered, but won't be connected

**Expected behavior**

Gateway should connect to the LYWSD03MMC sensor and report temperature

**Environment**

* OpenMQTTGateway V0.9.7

**Bug causes**

1. When device is first added to whitelist, `createOrUpdateDevice(mac, (isWhite ? device_flags_isWhiteL : device_flags_isBlackL), UNKNOWN_MODEL);` inside `updateWorB()` would create device with uninitialized `sensorModel` = `BEGINING` instead of `UNKNOWN_MODEL`.
2. During scan, `sensorModel` is only updated if it is `UNKNOWN_MODEL`, so it got never updated.
3. Even if `sensorModel` is `UNKNOWN_MODEL`, `createOrUpdateDevice(mac, device_flags_connect, LYWSD03MMC);` inside `process_bledata()` won't update `device->connect`, so the device is not connected.

**Possible fix**

It seems the following patch to `ZgatewayBT.ino` fixs this bug. Not sure if it has any side effect.

```
@@ -229,19 +229,22 @@ void createOrUpdateDevice(const char* mac, uint8_t flags, ble_sensor_model model
     strcpy(device->macAdr, mac);
     device->isDisc = flags & device_flags_isDisc;
     device->isWhtL = flags & device_flags_isWhiteL;
     device->isBlkL = flags & device_flags_isBlackL;
     device->connect = flags & device_flags_connect;
-    if (model != UNKNOWN_MODEL) device->sensorModel = model;
+    device->sensorModel = model;
     devices.push_back(device);
     newDevices++;
   } else {
     Log.trace(F("update %s" CR), mac);

     if (flags & device_flags_isDisc) {
       device->isDisc = true;
     }
+    if (flags & device_flags_connect) {
+      device->connect = true;
+    }

     if (model != UNKNOWN_MODEL) device->sensorModel = model;

     if (flags & device_flags_isWhiteL || flags & device_flags_isBlackL) {
       device->isWhtL = flags & device_flags_isWhiteL;
```

h2zero, 2021-09-12:
Thanks for this, the patch looks good to me and should resolve the issue brought up in #1032 as well.
Could you please submit this as a PR?

MountComb, 2021-09-19:
Glad to do so, but I have never submitted PR before. So please correct me if anything goes wrong.

closed 2021-09-19 by 1technophile