1technophile/OpenMQTTGateway #1034: Support for Rainpoint bL

garudaonekh, 2021-07-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

1technophile, 2021-07-24:
You need to check what data the sensor advertize with a mobile app like nRf Connect. Note that the sensor should be disconnected so as to be visible.
Write down or screenshot the hexadecimal values seen into nRF connect. Check supposed values and try to find correlation between them. You can also post here the screenshots of several scan and the expected moisture measurement.

1technophile, 2021-08-14:
Were you able to go further on this? I'm interested in this sensor as an alternative to Mi Flora.

garudaonekh, 2021-08-15:
Sorry, I try it but too many devices popup in nrfConnect and no clear pattern yet, maybe because too many devices nearby. i will try to find a place without any interference and try again

garudaonekh, 2021-08-15:
Yes, it is a good alternative. It is outdoor and cheaper price. I bought the sensor alone at 11$. But it has only moisture sensor.

1technophile, 2021-09-11:
Hi @garudaonekh,

Any progress on this?

You may also put OMG on an ESP32 and set to true these macros:
`pubBLEManufacturerData`
`pubBLEServiceUUID`

It would be easier to analyze the data on [MQTT explorer](http://mqtt-explorer.com/)

garudaonekh, 2021-09-12:
Sorry i am busy recently. will do it next week. I tried ot once but cannot find any specific pattern maybe due to too many Bluetooth devices nearby so no idea which one is which one. I used NRfconnect as you suggested

1technophile, 2021-12-07:
This sensor doesn't seem to have a Bluetooth component but rather rely on a gateway to convert its RF signals into Bluetooth

closed 2021-12-07 by 1technophile

garudaonekh, 2021-12-29:
I think so. So it work like most weather station which use RF chip. Do you have more detail about its RF chip and its spec?

NorthernMan54, 2021-12-29:
Did a bit of googling and rtl_433 has a decoder for this device https://github.com/merbanan/rtl_433/blob/master/src/devices/rainpoint.c

But it uses Manchester encoding which isn't supported by rtl_433_ESP https://github.com/merbanan/rtl_433/blob/9e6c8476748dabd274737d214fb6ce0d9b4c3f35/src/devices/rainpoint.c#L141

garudaonekh, 2021-12-29:
> Did a bit of googling and rtl_433 has a decoder for this device https://github.com/merbanan/rtl_433/blob/master/src/devices/rainpoint.c
>
> But it uses Manchester encoding which isn't supported by [...]

@NorthernMan54 Thanks for your info.
This RTL_433 is a stranger to me. However, I intent to use this sensor for my irrigation project. I have two questions:
1. Which chip is compatible for this?
2. What is the estimated distance it can communicate on open field.

Thanks;

NorthernMan54, 2021-12-29:
rtl_433 is another open source radio signal decoder package, and is quite popular in the community for receiving signals from temperature, weather and other RF sensor devices.  Full details are here - https://github.com/merbanan/rtl_433/blob/master/README.md

`Which chip is compatible for this?`

Devices like this are compatible https://www.amazon.ca/NooElec-NESDR-Mini-Compatible-ESD-Safe/dp/B00PAGS0HO/ref=sr_1_12?gclid=CjwKCAiAiKuOBhBQEiwAId_sK2roOBHuLy3JdbKccdUn6gC7Ut5KSSnKXXfAXWN1RU4XPM9XxoAlWhoCOZcQAvD_BwE&hvadid=538535012620&hvdev=c&hvlocphy=9000891&hvnetw=g&hvqmt=e&hvrand=13674738682221435229&hvtargid=kwd-350903603168&hydadcr=14279_10126889&keywords=usb+rtlsdr&qid=1640748091&sr=8-12

`What is the estimated distance it can communicate on open field ?`

This really depends on a lot of factors but the range of these is not that far, typically around 50 feet.  You would need to get one and see what works for you.

garudaonekh, 2021-12-29:
> rtl_433 is another open source radio signal decoder package, and is quite popular in the community for receiving signals from temperature, weather and other RF sensor devices. Full details are here [...]

[image]

NorthernMan54, 2021-12-29:
My distance numbers are based on my experience with Accurite and LaCrosse temperature and weather sensors, and with those I get those distances.  Hence the YMMV comment

One other thought, with the rtl_sdr dongle, you could drop a raspberry with the dongle installed into the middle of the green house, and back haul the signal over ethernet or wifi back to your main building.

garudaonekh, 2021-12-29:
Do we need to use RTL_SDR to receive the message? Can't we use normal RF receiver?
RTL_SDR is very costly around 20 to 40$ based on my recent price check.

NorthernMan54, 2021-12-29:
Based on my review of OMG, the pilight and rtl_433_ESP RF signal decoder's don't have support for this device ( based on grepping for 'rainpoint' in the the device decoders ).  But a quick check of rtl_433 shows that support was added in the the fall.  So if you want to start receiving signals now, you will need to use a rtl_sdr and rtl_433 on a RPI.

garudaonekh, 2021-12-30:
> Based on my review of OMG, the pilight and rtl_433_ESP RF signal decoder's don't have support for this device ( based on grepping for 'rainpoint' in the the device decoders ). But a quick check of [...]
--
`"So if you want to start receiving signals now, you will need to use a rtl_sdr and rtl_433 on a RPI."`
You are referring to the library or the hardware?

NorthernMan54, 2021-12-30:
That is the library and not hardware

garudaonekh, 2021-12-30:
ok, so based on rtl_433_ESP, we can use CC1101 transceiver to decode the data from this RainPoint, when the library is updated for it?

garudaonekh, 2021-12-31:
@NorthernMan54 I got my RainPoint soil moisture sensor working with my RTL_SDR today.
Anyhow, I need to make it works with rtl_433_ESP for my ESP32 project. Can you give me some suggestion on how to integrate this RapointPoint decoder into your rtl_433_ESP?

Thanks;

NorthernMan54, 2021-12-31:
With a bit of fiddling with platformio you should be able use a cloned copy of rtl_433_esp

then put the additional device decoder in the contrib directory and run the script in the tools directory called update_rtl_433_devices.sh

it will update all the various files except 1 to include the additional device decoder.  You need to update src/rtl_433_ESP.cpp with rtl_433_ESP.fragment".  You can ignore most of the rest of the comments, they are just reminders for me to manually update the various rtl_433 files in case of changes.

that will add it to the code base, build and upload it to your device, and be prepared for it to crash the first time it receives a signal.  Likely stack exhausted or something.  You will the need to work thru the decoder code to find the issue and resolve.

I had not included any devices that i did not support ask modulation or had a code pattern likely to cause an issue, and this one did. My rule of thumb was no code changes to the device decoders, to minimize maintaining the cloned device decoders.  As I didn’t have one, their was no urgency to port it

garudaonekh, 2021-12-31:
thanks for your detail explanation. I will try