1technophile/OpenMQTTGateway #1035: Can't get v0.9.7 to work

combatistor, 2021-07-22:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Hi everyone, I can't get v0.9.7 to work. I'm using wemos D1 mini esp32 board. I used the release from github, using arduino IDE to flash the board, using same setup as v0.9.6: manual wifi setup:
`#  define ESPWifiManualSetup true`
changed MQTT server address,
```
#ifndef MQTT_SERVER
#  define MQTT_SERVER "192.168.x.x"
```
 using BT module
`#define ZgatewayBT     "BT"       //ESP8266, ESP32`

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.7 release from github)

**Additional context**
On the serial monitor, after flashing it is stuck at 

```
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.7
```

I tried using the wifi manager but it doesn't create a hotspot
I might be missing something

I also had a compilation error because of `include <HTTPClient.h>` but I fixed it by updating ESP32 boards in the board manager

Thank you

1technophile, 2021-07-22:
Hi, did you tried to erase the flash before uploading and did you updated/added libraries with the ones given with the release?

combatistor, 2021-07-22:
Hi, Yes I cleaned-up the libraries with the updated ones from the release. What do you mean by erase the flash before uploading?
Is there a special command to do that?

combatistor, 2021-07-22:
I tried to upload an empty sketch before reuploading OMG, but same behavior:
```
N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.7

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.7

```

combatistor, 2021-07-22:
I tried using esp32dev-ble-libraries.zip and esp32dev-ble-cont-libraries.zip for the libraries, I also used OpenMQTTGateway_sources.zip for the arduino source code.
Still same issue (by the way what is the difference between the cont and other libraries?)

combatistor, 2021-07-22:
If I leave the line `// # define ESPWifiManualSetup true` commented
this is the logs I get:
```
I: 
************* WELCOME TO OpenMQTTGateway **************
I: OpenMQTTGateway Version: v0.9.7

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
I: 
************* WELCOME TO OpenMQTTGateway **************
I: OpenMQTTGateway Version: v0.9.7
{"mqtt_server":"192.168.1.17","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"your_password","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway"}*WM: [1] Free heap:        187408
*WM: [1] ESP SDK version:  v3.3.5-1-g85c43024c
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: secure
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: cert
*WM: [2] Added Parameter: name
*WM: [2] Added Parameter: topic
I: Attempting Wifi connection with saved AP: 0
I: BLE scans interval: 55555
I: BLE scans number before connect: 10
I: Publishing only BLE sensors: false
I: minrssi: 100
I: Low Power Mode: 0
I: OpenMQTTGateway modules: ["BT"]
I: ************** Setup OpenMQTTGateway end **************
W: MQTT connection...
W: MQTT client disconnected no BLE scan
W: failure_number_mqtt: 1
W: failed, rc=-2
W: failed, ssl error code=1073589320
etc...
```

PS: the mqtt server address is correct, but It doesn't create a hotspot or connect to my wifi

1technophile, 2021-07-23:
You need to erase the flash memory of the esp32 , so as to do this you can use the[ flash download tools](https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.8_0.zip) of ESPRESSIF.

As we have added an attribute related to secure connection to the flash memory it may be due to this. If yes I will update the release note

combatistor, 2021-07-23:
Thank you for providing the tool to flash the memory. after doing so and retrying to install OMG, this is what I get:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40183ce3  PS      : 0x00060330  A0      : 0x800d2077  A1      : 0x3ffcd6a0  
A2      : 0x00000000  A3      : 0x0000004e  A4      : 0x3ffba85c  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f219c  A9      : 0x3ffcd650  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffba844  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40183ce3:0x3ffcd6a0 0x400d2074:0x3ffcd6c0 0x400d2109:0x3ffcd710 0x400d5ea5:0x3ffcd730 0x400f38f2:0x3ffcd7e0 0x40090cfa:0x3ffcd800

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40183ce3  PS      : 0x00060330  A0      : 0x800d2077  A1      : 0x3ffcd6a0  
A2      : 0x00000000  A3      : 0x0000004e  A4      : 0x3ffba85c  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f219c  A9      : 0x3ffcd650  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffba844  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40183ce3:0x3ffcd6a0 0x400d2074:0x3ffcd6c0 0x400d2109:0x3ffcd710 0x400d5ea5:0x3ffcd730 0x400f38f2:0x3ffcd7e0 0x40090cfa:0x3ffcd800

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40183ce3  PS      : 0x00060330  A0      : 0x800d2077  A1      : 0x3ffcd6a0  
A2      : 0x00000000  A3      : 0x0000004e  A4      : 0x3ffba85c  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f219c  A9      : 0x3ffcd650  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffba844  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40183ce3:0x3ffcd6a0 0x400d2074:0x3ffcd6c0 0x400d2109:0x3ffcd710 0x400d5ea5:0x3ffcd730 0x400f38f2:0x3ffcd7e0 0x40090cfa:0x3ffcd800

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40183ce3  PS      : 0x00060330  A0      : 0x800d2077  A1      : 0x3ffcd6a0  
A2      : 0x00000000  A3      : 0x0000004e  A4      : 0x3ffba85c  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f219c  A9      : 0x3ffcd650  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffba844  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40183ce3:0x3ffcd6a0 0x400d2074:0x3ffcd6c0 0x400d2109:0x3ffcd710 0x400d5ea5:0x3ffcd730 0x400f38f2:0x3ffcd7e0 0x40090cfa:0x3ffcd800

Rebooting...
And so on...
```

combatistor, 2021-07-23:
Nevermind, after erasing and trying again, I got the same behavior
```
N: 
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: v0.9.7
```
and nothing else

combatistor, 2021-07-23:
Okay! after commenting the line 
`// # define ESPWifiManualSetup true`
It triggered the hotspot process, I'll configure using the hotspot then.

I think flashing the memory did solve the problem, but I think the `# define ESPWifiManualSetup true` might not trigger the proper routine in the code

1technophile, 2021-07-23:
I will try with Arduino IDE. I have it working one one board with `ESPWifiManualSetup true` from platformio

combatistor, 2021-07-23:
I can confirm it worked properly after setting up my wifi from the wemos AP, should I close the bug now or you'll check what is wrong with arduino IDE?

1technophile, 2021-07-25:
I tried to reproduce with Arduino IDE and ESPWifiManualSetup but was not able. All is working correctly here.
I'm closing the issue, if it appears again feel free to reopen

closed 2021-07-25 by 1technophile