1technophile/OpenMQTTGateway #1041: Strange behaviour during Read BLE characteristics, with 'pseudo' MiFlora appearing and accumulating, also causing reproducible crashes/restarts

DigiH, 2021-07-28:
To expain my current test setup:
I have several Playbulb Candles [(the old models)](https://www.mipow.com/collections/smart-light/products/playbulb-candle), which I was planning on integrating with the new Read/Write BLE characteristics possibility in OMG 0.9.7.

Figuring out the appropriate services and characteristics things looked good on my default BT OMG gateway with manually sending Read and Write requests to some candles. Wanting to be able to have Write commands take pretty immediate effect though, I decided to have a separate BLE OMG gateway, just for the Playbulb devices - which also helped a lot in pinning down the following described issues. 

I based the environment on the continous BT gatway, with the following variation

```
;esp32 BLE PLAYBULB
[env:esp32dev-ble-playbulb]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DGateway_Name="OMG_BLE_PLAYBULB"'
  '-DScanBeforeConnect=1'
  '-DTimeBtwRead=1000'
  '-DScan_duration=1000'
  ${otapass.build_flags}
  ${wifi.build_flags}
  ${mqtt.build_flags}
monitor_port = /dev/cu.SLAB_USBtoUART
upload_port = /dev/cu.SLAB_USBtoUART
upload_speed = 921600
```

I have also implemented a white-list with

{"white-list":["AA:BB:CC:DD:EE:FF"]}

as I don't really want/need any usual scan info for the candle devices.

Building, uploading - all went fine, but during more extensive testing, by starting to try and get the battery level for all nine candle devices, I found that after 4-6 device battery reads things slowed down, hung or aborted completely. While monitoring I experienced the following:

Initially, after uploading everything runs smoothly without any issues

```
…
N: BLE Connect begin
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:D1
N: Device detected: XX:XX:XX:XX:XX:F2
N: Device detected: XX:XX:XX:XX:XX:49
N: Device detected: XX:XX:XX:XX:XX:E6
N: Device detected: XX:XX:XX:XX:XX:E6
N: Device detected: XX:XX:XX:XX:XX:E6
N: Device detected: XX:XX:XX:XX:XX:C8
N: Device detected: XX:XX:XX:XX:XX:E0
N: Device detected: XX:XX:XX:XX:XX:02
N: Device detected: XX:XX:XX:XX:XX:0C
N: Device detected: XX:XX:XX:XX:XX:C7
N: Device detected: XX:XX:XX:XX:XX:EE
N: Device detected: XX:XX:XX:XX:XX:C2
N: Device detected: XX:XX:XX:XX:XX:EE
N: Device detected: XX:XX:XX:XX:XX:AB
N: Found 15 devices, scan number 27 end
N: BLE Connect begin
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:E6
N: Device detected: XX:XX:XX:XX:XX:F2
N: Device detected: XX:XX:XX:XX:XX:E6
N: Device detected: XX:XX:XX:XX:XX:C7
N: Device detected: XX:XX:XX:XX:XX:D1
N: Device detected: XX:XX:XX:XX:XX:AB
…
```

but as soon as I send the first Read request, to get the battery status for one candle, with

{"ble_read_address":"XX:XX:XX:XX:XX:49","ble_read_service":"180f","ble_read_char":"2a19","value_type":"HEX","ttl": 4}

the following happens

```
…
N: Device detected: XX:XX:XX:XX:XX:AB
N: Found 15 devices, scan number 53 end
N: BLE Connect begin
N: Subject: /BTtoMQTT/XXXXXXXXXX49
N: Received json : {"id":"XX:XX:XX:XX:XX:49","service":"0x180f","characteristic":"0x2a19","read":"64","success":true}
N: Subject: /BTtoMQTT/XXXXXXXXXX49
N: Received json : {"id":"XX:XX:XX:XX:XX:49","service":"0x180f","characteristic":"0x2a19","read":"64","success":true}
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:E6
…
```

with the Read response coming back nicely (twice, as discussed elsewhere), but with the additional perplexing

```
**N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic**
```

which looks like a MiFlora service. I do not have any MiFlora devices, and if one of my neighbours has any, which I very much doubt, why wouldn't they have shwon p previously. Or how else could this be explained?

This also continously stays now, with every Scan/Connect cycle

```
N: Device detected: XX:XX:XX:XX:XX:AB
N: Found 15 devices, scan number 85 end
BLE Connect begin
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:E6
…
N: Device detected: XX:XX:XX:XX:XX:AB
N: Found 15 devices, scan number 86 end
BLE Connect begin
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:E6
…
```

and with rhw next battery status READ request actually doubles up, so after the 3rd request I'm left with

```
N: Device detected: XX:XX:XX:XX:XX:AB
N: Found 15 devices, scan number 94 end
N: BLE Connect begin
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
N: BLE Connect end
N: Scan begin
N: Device detected: XX:XX:XX:XX:XX:E6
```

with the occasional `lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` added in.

Also, if a READ request seems to be timed just right a crash and reboot happens quite regularly

```
N: Found 15 devices, scan number 179 end
N: BLE Connect begin
N: Received json : {"ble_read_address":"XX:XX:XX:XX:XX:C7","ble_read_service":"180f","ble_read_char":"2a19","value_type":"HEX","ttl":4}
N: Failed to find service UUID: 00001204-0000-1000-8000-00805f9b34fb
N: Failed getting characteristic
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d869c  PS      : 0x00060c30  A0      : 0x800d8949  A1      : 0x3ffe2a70  
A2      : 0xbaad5678  A3      : 0x3ffe7464  A4      : 0x3ffc36e0  A5      : 0x3ffbd9c4  
A6      : 0x3ffc326c  A7      : 0x3ffe73dc  A8      : 0x800d8815  A9      : 0x3ffe2a50  
A10     : 0x3ffe2a7c  A11     : 0x3ffe7464  A12     : 0x3ffe7464  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad568d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d869c:0x3ffe2a70 0x400d8946:0x3ffe2ad0 0x40090d42:0x3ffe2af0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

```

I've now tried this with 3 different ESP32 devices, just to make sure it's not a flakey board which is causing this ;)

Any pointers to why this is happening, or am I trying to use the new READ/WRITE functionalty in a way it isn't intended?

Thanks

h2zero, 2021-07-28:
I just submitted #1042 that should fix this, please test if you can.

DigiH, 2021-07-28:
Looking good so far with initial testing AND the previously experienced duplicate responses have also disappeared :)

Thanks @h2zero

DigiH, 2021-07-28:
This also fixed the reappearing duplicate multiple recognitions during a single scan in 0.9.7 on my default BT OMG gateway, which I mentioned in https://github.com/1technophile/OpenMQTTGateway/issues/906#issuecomment-886388743, where I had to set `BLEScanDuplicateCacheSize` to `500` or even `1000`. All back to single recognitions with the default `200` value.

h2zero, 2021-07-28:
Great, thanks for testing :)

DigiH, 2021-07-28:
Thanks for fixing. Closing this as fixed with https://github.com/1technophile/OpenMQTTGateway/pull/1042

PR #1042 (2021-07-28): Bugfix: missing break in BLE connect switch.
Fixes #1041, GENERIC device case was falling through to MiFlora.

## Description:

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit a4a85ff1 (2021-07-28): Bugfix: missing break in BLE connect switch. (#1042)

Fixes #1041, GENERIC device case was falling through to MiFlora.
closed 2021-07-28 by DigiH