1technophile/OpenMQTTGateway #1052: Proposal for/query about a future Read/Write only mode/version for the BLE gateway

DigiH, 2021-08-04:
With the recent introduction of the Read/Write functionality for BLE devices I was very excited to try this out for my Playbulb Candle, which In have spread around my house. Currently I control them through one Raspberry Pi, but having several small ESP32s around the house would definitely improve the reachability.

While reading the battery status for each one of them is fine to only happen every few hours for me, and with a delay until the next connection happens after the next scheduled scan has finished, but writing ON/OFF changes or COLOUR changes to the candles should really happen instantaneously. I would expect the same for controlling most other BLE 'actuators, valves, lights, relays...'

So far, to test this possibility out a bit, I have quickly hacked a version with

• disabling the BLE Scan routine
• reducing m_connectTimeout to 500/1000 of the linked NimBLE library
• Setting '-DTimeBtwRead=0' and '-DScan_duration=100'

which gave me decent results in (almost) instantaneously controlling my candles, apart from very regular crashes/restarts of this hacked gateway ;)

For a future such version I would envision that connections only would happen when Read/Write MQTT messages are being received. Also, similarly to the current device detection, device profiles could also be included/added over time to be able to simplify commands.

I am aware that a version like this would be completely different to the current scanning, connecting, reporting workflow, and it would have to be running separately to any current OMG BLE installation, but it would also open the door to a vast amount of BLE devices which would want to be controlled promptly.

My questions here now are
• Would others also be interested in such a future Read/Write only version for immediate control?
• Is it worth pursuing this with OMG, or are there already other solutions which run on ESPs which can be easily distributed around the house for good BT coverage?
• Would anyone be interested in collaborating on this or have implementation suggestions, as I'm still very much a novice with the OMG code?  

Thanks for reading so far about my idea.

h2zero, 2021-08-04:
My advise for this would be to change the scan callback such that if a device is scanned that has pending read/write commands it would:

1. Record the remaining scan duration.
2. Connect immediately and read/write.
3. Restart the scan with the duration recorded in 1.

This would only have the effect of interrupting the scan process for a short time to facilitate the immediate read/write requirement while still completing the scan duration to record sensors etc. The only downside would be possibly missing a few advertisements during the connection period.

DigiH, 2021-08-04:
> My advise for this would be to change the scan callback such that if a device is scanned that has pending read/write commands it would:
>
>     1. Record the remaining scan duration.
>
>     2. Connect [...]

Wouldn't this also require continuous scanning with '-DTimeBtwRead=0' to avoid pauses before read/write commands are being executed? For my every day use a much higher TimeBtwRead value is usually fine, with it set to 0 only really necessary for BLE motion sensors and such, which want to be caught immediately.

That was my thinking behind avoiding scanning completely and to only execute read/write commands with such a version.

h2zero, 2021-08-04:
I order to connect to the device it's best to scan for it first so that the correct address can be obtained, or more specifically the address type. Many people who try to connect to a device without scanning get stuck because the address type is not `0`. For your use case it may well be fine, I just wouldn't want others to do the same with different devices and wonder why it's not working.

The adjustment you made to the library is not necessary, you can call the API function from OMG instead to change that value: https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_client.html#a4068b29a9bd12e3110465908864dd20e

DigiH, 2021-08-04:
> The adjustment you made to the library is not necessary, you can call the API function from OMG instead to change that value: https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_client.html#a4068b29a9bd12e3110465908864dd20e

Thanks a lot for that, it'll make my testing a lot less hacky to be able to call it directly from OMG :)

I also get the address issue, with best still scanning for assuring the correct address type, so will make some changes here for further testing with your advice.

1technophile, 2021-08-04:
+1 for immediate BLE connection + write, I think it makes sense as a new feature

DigiH, 2022-01-19:
I suppose with @h2zero's WIP - [Add immediate parameter to BLE connection commands](https://github.com/1technophile/OpenMQTTGateway/pull/1137) - this is being addressed there and can happily be closed here. Thanks

closed 2022-01-19 by DigiH

h2zero, 2022-01-20:
@DigiH, I would appreciate your testing if you're interested. I just updated the PR to what I believe to be fully functional.