1technophile/OpenMQTTGateway #1055: DHT sensor reading unreliable

akemper, 2021-08-08:
Hi,

according to my experiences with OMG 0.9.5, 0.9.6 and latest version, reading of (even different) DHT 22 sensors from ESP32 board is quite unreliable, i.e. even when using the just few cm long jumper cable shipped with the sensor there are sporadic "failed to read the sensor" messages. The readout gets significantly worse when enabling ADC and / or other sensor inputs. I believe it could be a timing / interrupt issue of the OMG, but so far have no clue where to start fixing this. Any related ideas / experiences on this?

Thx
Andreas

1technophile, 2021-08-08:
Hi,

Could you share your configuration especialy the time between readings used ?

akemper, 2021-08-09:
I did not touch anything in config_DHT.h, except for temporary lowering TimeBetweenReadingDHT from default 30000 down to 5000, but this didn't make any obvious change, apart from more frequent readings. In platformio.ini added corresponding module and library like this:

`[env:esp32dev-rf]
platform = ${com.esp32_platform}
board = esp32dev
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.rc-switch}
  ${libraries.dht}
  ${libraries.unifiedsensor}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
  '-DZsensorDHT="DHT"'
  '-DGateway_Name="OpenMQTTGateway_ESP32_RF"'`

and in User_config.h just uncommented #define ZsensorDHT "DHT", plus added wifi config and MQTT server IP.

Thx
Andreas

1technophile, 2021-08-11:
Hi, COuld you try the linked PR, I have updated the libraries. Hopefully it increases the reliability of the readings.

1technophile, 2021-08-16:
@akemper would you be able to test this?

akemper, 2021-08-17:
Sorry, I didn't post the answer here but erroneously wrote back to the notifications email. 

I changed the two libraries as recommended and flashed the ESP32 again. Afterwards reading got even much worse, i.e. apart from more failures all read-out values were largely incorrect.

Thus I flashed it back to the original version, but this time connected the DHT22 to 3.3V instead of 5V power supply, as recommended on https://docs.openmqttgateway.com/setitup/sensors.html#compatible-sensors

Afterwards it is working very well. Even for some meters of (shielded) cable the sensor is connected with and just 5s read-out interval. At least during the last ~ 9h no reading failed and the ESP32 is overall working stable.

closed 2021-08-29 by 1technophile