1technophile/OpenMQTTGateway #1063: WT32-ETH01 support

mgx0, 2021-08-16:
**Is your feature request related to a problem? Please describe.**
no, just a feature request - support of new device

**Describe the solution you'd like**
support WT32-ETH01 board from Wireless-Tag

**Describe alternatives you've considered**
build for esp32-olimex-gtw-ble-eth (the only ethernet-enabled device with OMG support) does not run on this board, therefore I am opening this request.

**Additional context**
ESP_Easy_mega_20210802_normal_ESP32_4M316k_ETH-factory.bin works properly on this board, ethernet is working as it should

Hi, I came across these modules and I like they have Ethernet. I'd like to have a BLE>Ethernet interface from this thingy. Do you think you can make OMG run on this board please? If I'm not mistaken, it does not run on this yet. If I'm wrong, could you please direct me what to upload to it?

thanks

mgx0, 2021-08-16:
Here is the datasheet: http://www.wireless-tag.com/wp-content/uploads/2020/08/WT32-ETH01规格书V1.2EN（英文）.pdf

Looks like it uses the same Lan chipset as Olimex POE/gtw

1technophile, 2021-08-16:
Could you try to replace [here](https://github.com/1technophile/OpenMQTTGateway/blob/eaa7bc94b241f27123b36824ceb7b478f4b94e60/main/main.ino#L199)

`#    define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT`
by
`#    define ETH_CLK_MODE  ETH_CLOCK_GPIO0_IN`

and here
`#    define ETH_PHY_POWER 12`
by
`#    define ETH_PHY_POWER 16`

mgx0, 2021-08-17:
thanks, I have tried that. The Ethernet LED blinks now, I have enabled TRACE logging and all I can see on uart is this (with NetworkAdvancedSetup false)
[image]

this is with NetworkAdvancedSetup true (boot loop, this repeats in loops):
```
************* WELCOME TO OpenMQTTGateway **************
N: OpenMQTTGateway Version: version_tag
T: Adv eth cfg
E (11) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x401658ef on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008fc38:0x3ffcda10 0x4008feb5:0x3ffcda30 0x401658ef:0x3ffcda50 0x400dfb63:0x3ffcda90 0x400d3961:0x3ffcdae0 0x400d7f5b:0x3ffcdb40 0x40104302:0x3ffcdbf0 0x40090eb6:0x3ffcdc10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec
```

I have used bootloader and boot_app0 from the guide here: https://docs.openmqttgateway.com/upload/binaries.html#esp32

and here are the settings used while uploading
[image]

have I done something wrong? this procedure worked fine with ESP32-dev, but I never used my own compiled ones

mgx0, 2021-08-17:
looks like it's solved, can you add it to the builds or wiki:

in order to use OMG on WT32-ETH01 boardie you need to adjust in main.ino

#  ifdef ESP32_ETHERNET
**#    define ETH_CLK_MODE  ETH_CLOCK_GPIO0_IN
#    define ETH_PHY_POWER 16
#    define ETH_PHY_ADDR 1**   //this is what has been missing, it was 0 before
#    include <ETH.h>

closed 2021-08-17 by mgx0

1technophile, 2021-08-17:
Thanks for the feedback, do you know if there is a LED available for user programs? I see ones for power rx and tx on the doc but not for user usage.

mgx0, 2021-08-17:
I believe not, there is a power led (unmanaged), then RX and TX LEDs but bound directly to the UART, so the only option is to pull up/down the uart and not using it. Can I test something for you?

1technophile, 2021-08-17:
I think it would be OK like this for the LED. If you could verify that the new environment into the PR is working it would be great

mgx0, 2021-08-17:
if you give me instructions what to do, I'll be glad to help. what should i do with the led and what should I verify?

btw. should I open a new issue for this or shall we continue here?

1technophile, 2021-08-17:
Do you use platformio?
If not you can follow these [instructions](https://docs.openmqttgateway.com/upload/builds.html#introduction) - Configure & Upload with PlatformIO

Instead of taking the code from the development branch you take it from the [PR](https://github.com/1technophile/OpenMQTTGateway/pull/1064)
Remove the ; in front of `;default_envs = esp32-wt32-eth01-ble-eth`
And you can build from platformio

If you don't want to go through this learning curve I can ask you to test when beta binary files will be available.

mgx0, 2021-08-17:
yes I use platformio, I'll do it within few hourss

mgx0, 2021-08-17:
done, thanks a lot 😁

DangerD1024, 2022-04-16:
Can use this board for RF and IR?

1technophile, 2022-04-16:
You would need to check that the pin used for IR and RF components are available