1technophile/OpenMQTTGateway #1067: CC1101 not transmitting nor receiving

mhtrinh, 2021-08-23:
I am using :
- development branch
- CC1101 :  https://www.aliexpress.com/item/33059343309.html
- ESP32-WROOM-32D DevkitC_v4
- Power by USB. Power the module via the 3v3 pin on the ESP32 : is it how it supposed to be ?
- Check all the wiring according to https://docs.openmqttgateway.com/setitup/rf.html#cc1101-pinout
- Build with : 
```
[env:esp32dev-cc1101]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.rc-switch}
  ${libraries.smartrc-cc1101-driver-lib}
;   ${libraries.rtl_433_ESP}
;   ${libraries.esppilight}
;   ${libraries.newremoteswitch}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
;   '-DZgatewayRF2="RF2"'
;   '-DZgatewayRTL_433="RTL_433"'
;   '-DZgatewayPilight="Pilight"'
  '-DZradioCC1101="CC1101"'
  '-DGateway_Name="OpenMQTTGateway_multi_receiver"'
  '-DBase_Topic="CC1101/"'
;   '-DvalueAsASubject=true'  ; mqtt topic includes model and device (rtl_433) or protocol and id ( RF and PiLight )
;  '-DDEFAULT_RECEIVER=1'  ; Default receiver to enable on startup
  '-DESPWifiManualSetup'
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'

  ${cred.build_flags}
```
I have a second ESP32 than can receive: checked with a RF door sensor. 
The ESP32 with CC1101 : cannot receive anything, even at 1cm range. 
Any suggestion ?

NorthernMan54, 2021-08-23:
Can you share the serial output from the device during startup and testing ?

In my devices I have powered the cc1101 via the 3v3 pin on the esp32 board, and it works well.

mhtrinh, 2021-08-23:
```
> Executing task: platformio run --target upload --target monitor --environment esp32dev-cc1101 <

Processing esp32dev-cc1101 (platform: espressif32@3.3.1; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Converting main.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.8.0
|-- <ArduinoJson> 5.13.4
|-- <ArduinoLog> 1.0.3+sha.d13cd80
|-- <WifiManager> 1.0.0+sha.c3ff582
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <rc-switch> 2.6.2+sha.36dfb45
|-- <SmartRC-CC1101-Driver-Lib> 2.5.5
|   |-- <SPI> 1.0
|-- <ESP32 BLE Arduino> 1.0.1
|-- <Wire> 1.0.1
|-- <ArduinoOTA> 1.0
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <EEPROM> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Preferences> 1.0
|-- <SPI> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <Update> 1.0
Building in release mode
Compiling .pio/build/esp32dev-cc1101/src/main.ino.cpp.o
Retrieving maximum program size .pio/build/esp32dev-cc1101/firmware.elf
Checking size .pio/build/esp32dev-cc1101/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.2% (used 46376 bytes from 327680 bytes)
Flash: [=====     ]  50.6% (used 994752 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
[... 136 lines cut ...]
T: Min ind: 0
T: store code : 9610174 / 52766
T: Col: val/timestamp
T: mem code : 9610174 / 52766
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
T: mem code : 0 / 0
N: Received json : {"value":9610174,"protocol":34,"length":24,"delay":326,"avgError":82.08334}
```

I have another OMG running on `test/OMG` with RF gateway. That would explain : `Hey I got a callback test/OMG/433toMQTT`
But is it normal that this one track the other one ??

Some setup picture: 

https://github.com/mhtrinh/wiki/blob/master/images/20210824_111329.jpg
https://github.com/mhtrinh/wiki/blob/master/images/20210824_111205.jpg

NorthernMan54, 2021-08-24:
Is your wiring accurate for that cc1101 version, I can't tell from your pictures?  I found that the wiring was different with that version, and after alot of googling found this data sheet https://github.com/NorthernMan54/rtl_433_ESP/blob/fsk/E07-M1101D-TH_Usermanual_EN_v1.30.pdf

_I have another OMG running on test/OMG with RF gateway. That would explain : Hey I got a callback test/OMG/433toMQTT
But is it normal that this one track the other one ??_

Yes its a feature of the OMG package, and it is meant to stop duplicate broadcasting of received messages.

PS I just tested RF receive with the latest development package and my SONOFF RF remote on 433.92 was received ( Mine has the RF Module and the rtf_433 Module )

mhtrinh, 2021-08-24:
I just took some better photos: https://github.com/mhtrinh/wiki/wiki/Photo-CC1101

Your CC1101 do look very similar to mine. The pin do seems to match the aliexpress page. 

So either my eyes are cross, either the module is dead on arrival, which I don't really know how to check it :-( 

PS: I have this weird and random thing happening : when I have the module plugged to the ESP32, sometime PlatformIO failed to upload with mismatch MD5. TO solve that, I need to unplug the 3v3 pin and upload again.
The mismatch error happen like 50% of the time. 
Powersupply issue ?? Currently power by USB from the computer.

NorthernMan54, 2021-08-24:
With mine I have not seen that issue, but I have added some filtering caps to my circuit

https://github.com/NorthernMan54/rtl_433_ESP/blob/fsk/docs/CC1101%20with%20BME280%20and%20BH1750.png

I’m wondering if there is a library that can validate the board and wiring for you? With the RF module, the board isn’t really exercised.

mhtrinh, 2021-08-24:
My chip is : 
CC1101 
94M1T1G
0914

Nor sure if that matter ...

NorthernMan54, 2021-08-24:
Just to check out the OMG build etc, I compiled the current development branch with your settings and applied it to one of my devices.  Success

With my device I wired it differently and need to manually specify the connections.

```
[env:issue_1067]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.rc-switch}
  ${libraries.smartrc-cc1101-driver-lib}
;   ${libraries.rtl_433_ESP}
;   ${libraries.esppilight}
;   ${libraries.newremoteswitch}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
;   '-DZgatewayRF2="RF2"'
;   '-DZgatewayRTL_433="RTL_433"'
;   '-DZgatewayPilight="Pilight"'
  '-DZradioCC1101="CC1101"'
  '-DGateway_Name="OpenMQTTGateway_multi_receiver"'
  '-DBase_Topic="CC1101/"'
;   '-DvalueAsASubject=true'  ; mqtt topic includes model and device (rtl_433) or protocol and id ( RF and PiLight )
;  '-DDEFAULT_RECEIVER=1'  ; Default receiver to enable on startup
;  '-DESPWifiManualSetup'
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
; My settings  
  '-DRF_EMITTER_GPIO=13'  ; GDO0 - Breadboard is 22 versus 13 on soldered
  '-DRF_RECEIVER_GPIO=4'  ; GDO2
  '-Dota_password=""'
  '-DMDNS_SD=true'
  '-DESPWifiManualSetup=true'
  '-DUSE_MAC_AS_GATEWAY_NAME'
  '-Dwifi_ssid1="xxxxxx"'
  '-Dwifi_ssid="xxxxxx"'
  '-Dwifi_password="xxxxxx"'
  '-Dwifi_password1="xxxxxx"'
  '-DSERIAL_BAUD=921600'

monitor_port = /dev/cu.SLAB_USBtoUART
monitor_speed = 921600
;upload_port = /dev/cu.SLAB_USBtoUART
;upload_speed = 921600

upload_port = OMG_3C71BF9E0770.local 
; OMG_3C71BF9E0770.local - Family Room aaaaaa
; OMG_240AC4EF5C0C.local - Floater bbbbbb
; OMG_9C9C1FC6F56C.local - Cottage cccccc

upload_protocol = espota
upload_flags =
  --port=8266
```

This was received from my SONOFF RF Remote
```
{"value":15467666,"protocol":1,"length":24,"delay":243,"mhz":433.92}
```

NorthernMan54, 2021-08-24:
This is my built device

[image]

mhtrinh, 2021-08-24:
I wonder if my chip is a fake one with different pinout or simply a different version from TI

@NorthernMan54 what kind of range do you get ? How many wall can it go through ? 
My main reason for ordering the CC1101 was hoping for range and not muck around with antennas
If it have good range, I may order another one (the main cons is the weeks of waiting for it to arrive)

NorthernMan54, 2021-08-25:
I find the range is not a good as a RTL_SDR, about 60% to 70% of the range I get.  And less than my rxb6 as well.  But overall still usable for me.  My main use case is rtl_433 for my temperature sensors and controlling my RF devices, which works well for me.

closed 2022-04-16 by stale[bot]

rantwolf, 2023-01-31:
@NorthernMan54
Can you tell where you bought your cc1101 module? I got mine yesterday from amazon. Found also the pinout at the ebyte datasheet pdf. After flashing the openmqttgateway firmware to my esp32 wroom and connect to wifi/mqtt I received some data from my weatherstation. This morning the module only received data sporadically. There were a few error messages in the broker: "unknown". What the heck is that?
[image]

Now the module would not receive data anymore...

NorthernMan54, 2023-02-01:
Mine came from Ali Express a few years ago.

The unknown messages, are from signals that were received, but failed parsing.  It could be an unknown device, or a noisy signal or random noise or ???

I find that the signal reception range is a bit limited compared to a rtl_sdr, and if the antenna is vertical reception is better.

rantwolf, 2023-02-06:
So I bought a new CC1101 transceiver. But new problems...
The CC1101 recieves messages but all of this messages are 'unknown' and the transmitter of the weatherstation is about 50cm away from the CC1101 module.
I think there is a problem inside the esp32_rtl_433 image.
Now I made a news test.
Plug an additional RTL_433 USB device into my pc and run rtl_433_sdr.
This piece of software gives me all needed information about my weather station.
[image]

So...how do I find the issue in the esp32?

NorthernMan54, 2023-02-06:
The Fineoffset device may not be supported. Looking at the device decoder, there appears to be different models one with OOK which should be okay, and one with FSK modulation that isn't.  Do you know which one you have ?

dreamcat4, 2023-03-01:
> I find the range is not a good as a RTL_SDR, about 60% to 70% of the range I get. And less than my rxb6

hello, i was curious how to get best range for 433mhz, and came across your commentary here. however i am still unclear what you mean by:

rtl_sdr - do you mean the 'RTL-SDR Blog V3' ? (or an earlier version), and with what software?

i also happen to have rxb6 and sx1278 modules here, can i connect them to these generic ESP-WROOM-32 modules and flash your 1technophile/OpenMQTTGateway?

It would be nice to try all 3 options (ideally). However if not, i can still buy something else if there is any better suggestion right? For example on your website it says for 433mhz is marked 'preferred' (or 'recommended') the seperate srx/stx 822 superhet 433 individual Rx and Tx modules. Would that be instead better for best 433mhz listening signal strength here for these openmqttgateway/esp32 based solution? (that isn't rtl-sdr). Or to purchase instead the TTGO with LORA?

Sorry if that is a bit much to be asking... perhaps you cannot give experience with all of those mentioned products. Any helps at all is really appreciated. So much thanks

DigiH, 2023-03-01:
Hi @dreamcat4,

while @NorthernMan54 might be able to give even more insight, what kind of 433Mhz devices are you trying to receive?

It needs to be differentiated between the 'standard' 433Mhz devices which can be received with RF/RF2 build gateways, or specifically rtl_433_ESP devices, which do require a CC1101, SX1278 module or the built in SX127x of a Heltec or LilyGo LoRa board.

For my set-up the recommended separate srx/stx 822 superhet 433 individual Rx and Tx modules have the best reception/transmission for a 'standard' RF 433Mhz gateway …

While for an rtl_433_ESP gateway I get the best results with my LilyGo TTGO LoRa32-OLED v2.1.6.

So it really depends which kind of 433Mhz gateway you want to set up.

dreamcat4, 2023-03-01:
ok thanks for the help! it is for listening to random commercial 433mhz devices around the house

based on your reply i ordered:

* 1x Nooelec RTL-SDR v5 SDR
* 2x (pair of)  srx/stx 822 superhet 433 individual Rx and Tx modules

sp i suppose then my only main question left over is can i flash this openmqttgateway onto esp-wroom-32 module. which is just some generic esp32 module. of course, that is not the main topic of this issue, (and those srx/stx 822 superhet modules isnt going to arrive from china for some week, to try them)

DigiH, 2023-03-01:
Please be aware that the RTL-SDR will only work with a Raspberry Pi or PC, not with any ESP module.

Your esp-wroom-32 will be fine for OpenMQTTGateway and connecting the srx/stx 822.

dreamcat4, 2023-03-01:
yep! all fine (got several rpis here already). & thanks for clarifying about the eesp-wroom-32