1technophile/OpenMQTTGateway #1073: Support for new Mi soil sensor MAX version

garudaonekh, 2021-09-08:
hi,
please add support for new outdoor version of Mi Flora soil sensors. 
https://a.aliexpress.com/_mL15QNt[MAX version](https://a.aliexpress.com/_mL15QNt) 

I got one. and here is its NRF Connect specs

0x020106030295FE141695FE712098006734786C8D7CC40D09100200000C09466C6F7765722063617265

garudaonekh, 2021-09-08:
[image]
[image]

1technophile, 2021-09-08:
Checking the service data on the screenshot format it should be compatible with the all the OMG versions that support Mi Flora, did you try already?

1technophile, 2021-12-07:
Already compatible

closed 2021-12-07 by 1technophile

garudaonekh, 2022-01-13:
@1technophile Sorry, I was confused with another Mi Flora. Actually, it has some change in compared with the Mi Flora.
Please take a look at this: https://github.com/custom-components/ble_monitor/issues/43