1technophile/OpenMQTTGateway #1075: IR Signal does not get sent

spacemanspiff2007, 2021-09-12:
Hello,

I'm having trouble getting my IR transmitter working (OMG 0.9.8).
I checked with my mobile phone and the IR-LED is lighting up when I send a command.
I checked with a remote and I can receive signals with the receiver.
However I am not able to re-receive the transmitted Signal (emitter -> receiver).

In the terminal I see the following message when I publish a command

```
N: Subject: /IRtoMQTT
N: Received json : {"value":1270235327,"protocol":3}
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081C78 arg=3FFBDECC
N: Received json : {"value":1270235327,"protocol":3}
```

It looks like there is an error which prevents the whole IR signal to be transmitted but I am not sure.
Is my assumption correct or is the error somewhere else and I have to keep looking?
Thank you for your help!

1technophile, 2021-09-12:
Hello,

Could you share the command that you are publishing to the MQTT broker so that we can reproduce it?

spacemanspiff2007, 2021-09-12:
I am using this topic ( I am using another prefix through the build flags)
```
mqttgw/OpenMQTTGateway_ESP32_IR/commands/MQTTtoIR
```
with this payload:
```json
{"value":551489775}
```

Build Flags (without wifi creds):
```
build_flags =
  ${com-esp.build_flags}
  '-DGateway_Name="OpenMQTTGateway_ESP32_IR"'
  '-DZgatewayIR="IR"'
  '-DZsensorBME280="BME280"'
  '-DBase_Topic="mqttgw/"'
  '-UsimpleReceiving'                ; Disable HA discovery
  '-UZmqttDiscovery'
```

1technophile, 2021-09-12:
Could you add also the received signal json please?

spacemanspiff2007, 2021-09-12:
What do you mean? I don't receive anything when I point the transmitter to the receiver.
Or do you mean what the receiver receives when I point my TV remote on it?

1technophile, 2021-09-12:
> Or do you mean what the receiver receives when I point my TV remote on it?

Exactly, if you could share this json please

spacemanspiff2007, 2021-09-13:
This is the log when I hit a button on the remote
```
N: Subject: /IRtoMQTT
N: Received json : {"value":1270267967,"protocol":3,"bits":32,"hex":"0x4BB6C03F","protocol_name":"NEC","raw":"8958,4522,544,574,546,1698,548,572,546,576,544,1698,546,572,546,1698,546,1696,546,1694,546,574,548,1696,544,1696,546,572,546,1696,546,1696,550,568,548,1694,548,1694,572,548,572,546,546,576,548,570,572,550,568,548,548,572,574,546,548,1696,544,1698,546,1694,546,1696,546,1696,570,1674,546"}
N: no pub. dupl
N: Received json : {"value":1270267967,"protocol":3,"bits":32,"hex":"0x4BB6C03F","protocol_name":"NEC","raw":"8958,4522,544,574,546,1698,548,572,546,576,544,1698,546,572,546,1698,546,1696,546,1694,546,574,548,1696,544,1696,546,572,546,1696,546,1696,550,568,548,1694,548,1694,572,548,572,546,546,576,548,570,572,550,568,548,548,572,574,546,548,1696,544,1698,546,1694,546,1696,546,1696,570,1674,546"}
```

Sending ```{"value": 1270267967}``` to ```mqttgw/OpenMQTTGateway_ESP32_IR/commands/MQTTtoIR```

```
N: Received json : {"value":1270267967}
N: Sending IR signal with
N: MQTTtoIR OK
N: Subject: /IRtoMQTT
N: Received json : {"value":1270267967}
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081C78 arg=3FFBDECC
N: Received json : {"value":1270267967}
```

1technophile, 2021-09-14:
Could you try to send with `hex` format instead of `value`

{"hex":"0x4BB6C03F","protocol_name":"NEC"}

spacemanspiff2007, 2021-09-14:
Here is the log
```
N: Received json : {"hex":"0x4BB6C03F","protocol_name":"NEC"}
N: Sending IR signal with NEC
N: MQTTtoIR OK
N: Subject: /IRtoMQTT
N: Received json : {"hex":"0x4BB6C03F","protocol_name":"NEC"}
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081C78 arg=3FFBDECC
N: Received json : {"hex":"0x4BB6C03F","protocol_name":"NEC"}
```

A strange observation I made:
Even if I shield the transmitter the value I am sending to MQTTtoIR is automatically resend under IRtoMQTT.
Is this the desired behavior?

1technophile, 2021-09-14:
> resend under IRtoMQTT.
Is this the desired behavior?

Yes this is an acknowledgment

When using the hex format do you get a better result?

spacemanspiff2007, 2021-09-14:
> When using the hex format do you get a better result?

I am pointing the transmitter directly onto the receiver but I still only get the acknowledgment.

Do you think this line indicates that there is an issue or can this line be ignored?
```
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081C78 arg=3FFBDECC
```

1technophile, 2021-09-14:
> Do you think this line indicates that there is an issue or can this line be ignored?

Seems related to https://github.com/crankyoldgit/IRremoteESP8266/pull/1435

Can you try to update 
`irremoteesp = IRremoteESP8266@2.7.15`

to

`irremoteesp = IRremoteESP8266@2.7.16`

in [platformio.ini](https://github.com/1technophile/OpenMQTTGateway/blob/ed4949f7693dea17727269cfc0d6f9ea1f99b64f/platformio.ini#L93)

spacemanspiff2007, 2021-09-15:
I did (clean and rebuild) as you suggested but the message is still there.

```
Dependency Graph
|-- <IRremoteESP8266> 2.7.16
```

```
N: Received json : {"value":1270267967}
N: Sending IR signal with
N: MQTTtoIR OK
N: Subject: /IRtoMQTT
N: Received json : {"value":1270267967}
[E][esp32-hal-cpu.c:93] addApbChangeCallback(): duplicate func=40081C78 arg=3FFBDECC
N: Received json : {"value":1270267967}
```

closed 2022-01-09 by spacemanspiff2007