1technophile/OpenMQTTGateway #1076: new BLE Soil Moisture Sensor

garudaonekh, 2021-09-13:
Hi,
How to add support for this soil sensor? It's very good quality and it include EC, Moisture, Light and Temperature.

[image]
[image]

Here's its raw data from NrfConnect:

0x02010503198001050212180F1807FF453B9D38C1A40B0941692D5468696E6B6572

Thanks;

1technophile, 2021-09-13:
Hi,

You need to add also the values seen on the app along as the raw data.
Ec= ? 
Temperature= ? 
Moisture= ? 
Lux= ? 
Screenshots showing raw data or detailled data (manufacturer data, service data, service data uuid )

At leat for 3 set of values.

1technophile, 2021-09-14:
Do you have a link to this sensor by the way?

garudaonekh, 2021-09-14:
Its app is in Chinese. I cannot figure out how to use it yet.

garudaonekh, 2021-09-14:
[image]
now I am testing

garudaonekh, 2021-09-14:
Biggest one is Mi Max version, medium one is the one I mentioned here. smallest one is Mi Flower care

garudaonekh, 2021-09-14:
[image]
[image]
Here're the two test cases.From its app and NrfConnect

1technophile, 2021-09-14:
> Here're the two test cases.From its app and NrfConnect

I see on the app that the values are changing but unfortunately the value seen on NRFconnect are constant, Raw data and Details are the same between the 2 screenshots.
Could you verify if the Raw data and Details are changing between several measurements?

h2zero, 2021-09-14:
Breaking down the raw data you have:
`02 01 05` = Advertising flags, (BLE_HS_ADV_F_DISC_LTD | BLE_HS_ADV_F_BREDR_UNSUP) which is interesting, it's using rapid advertising here, good way to burn up batteries.

`03 19 8001` = Appearance is type 0x0180, Generic Remote Control

`05 02 12180F18` = Incomplete service uuids, 16bit, 0x1812, 0x181F.

`07 FF 453B9D38C1A4` = Manufacturer data, this is where you'll find the values you're looking for.

`0B 09 41692D5468696E6B6572` = Device name, "Ai-Thinker"

1technophile, 2021-09-14:
> Biggest one is Mi Max version, medium one is the one I mentioned here. smallest one is Mi Flower care

Do you have a link to the one at the middle (shopping link), I tried HJPSWLO2 but no chance

garudaonekh, 2021-09-15:
@1technophile When I click on "Recheck" in the app, NRFConnect ask me "Do you want to debug the connection" thus it looks like the device needs wait for connection from app. And when I click on debug, i saw these value change in "Clinet" section of NRFConnect
[image]
[image]

garudaonekh, 2021-09-15:
> > Biggest one is Mi Max version, medium one is the one I mentioned here. smallest one is Mi Flower care
> 
> Do you have a link to the one at the middle (shopping link), I tried HJPSWLO2 but no chance

It's not available on AliExpress yet.

1technophile, 2021-09-17:
@garudaonekh 
> When I click on "Recheck" in the app, NRFConnect ask me "Do you want to debug the connection" thus it looks like the device needs wait for connection from app. And when I click on debug, i saw these value change in "Clinet" section of NRFConnect

This page is the result of the device connection. As a first attempt we can look of the broadcasted data (easier)

THe issue that I have with the 2 screenshots given is that they have the same raw values, if you take a second look to you see these raw value changing depending on the temperature or other sensor values?

garudaonekh, 2022-01-12:
Checked with the manufacturer. The data is encrypted and need to be decrypted from their cloud server and send back to app.

closed 2022-01-12 by garudaonekh

h2zero, 2022-01-12:
>The data is encrypted and need to be decrypted from their cloud server and send back to app.

That's insane, another app for someone to monitor users telemetry data, using encryption to force the app use? Sorry, that irks me, makes me want to hack it just to spite them lol. Do you have a link to order these?

garudaonekh, 2022-04-19:
> That's insane, another app for someone to monitor users telemetry data, using encryption to force the app use? Sorry, that irks me, makes me want to hack it just to spite them lol. Do you have a [...]

It's sold on Taobao.com only. You can buy it only if you have shipping agent in China(they only ship within China).