1technophile/OpenMQTTGateway #1083: Add INKBIRD IBBQ IBT-2X BBQ temperature sensor

freddyf, 2021-10-01:
This probe sensor is already recognized but not in the right way.

It is marked as an IBT-4XS with 4 thermometers, but this model has only two.
By the way, the 2 thermometers works fine labeled using tempc and tempc2

Thank You & great work!

1technophile, 2021-10-01:
Thanks, Could you share a screenshot with the manufacturer data of it like in the discussion [here]

freddyf, 2021-10-02:
[image]

here it is.
Thank you again!!

1technophile, 2021-11-04:
@freddyf could you try out this branch please?
https://github.com/1technophile/OpenMQTTGateway/tree/decoder-dev

And see if the model is correctly recognized

1technophile, 2021-12-05:
Done

closed 2021-12-05 by 1technophile