1technophile/OpenMQTTGateway #1095: Please add PC-60FW Fingertip Oximeter

yeuking, 2021-11-03:
[image]

**Is your feature request related to a problem? Please describe.**
No, this is new device.

**Describe the solution you'd like**
Just add support read info for battery, Sp02, PR and wave.

**Describe alternatives you've considered**
None.

1technophile, 2021-11-03:
Do you see manufacturer data like below if you wait:
[image]

1technophile, 2021-11-03:
Some usefull resources:
https://github.com/pythag/pc-60fw

yeuking, 2021-11-04:
> 
> 
> Do you see manufacturer data like below if you wait: [image]

Sorry, can't see manufacturer data like above.

yeuking, 2021-11-04:
> 
> 
> Some usefull resources: https://github.com/pythag/pc-60fw

Thanks, I found another usefull resources as below (more detail for this device) :
https://anaesthetics.app/blog/posts/2020/bluetooth/

closed 2022-04-16 by stale[bot]