1technophile/OpenMQTTGateway #1097: Please add Unbrand Smart Scale

yeuking, 2021-11-04:
[image]
[image]

Is your feature request related to a problem? Please describe.
No, this is new device.

Describe the solution you'd like
Just add support read info for battery, weight and other if possible like below:-
[image]

Describe alternatives you've considered
None.

1technophile, 2021-11-05:
Could you take another screenshot of nrfConnect please with some weight on it, along with the corresponding weight (different from the one displayed above)?
This would enable to confirm the parsing

yeuking, 2021-11-06:
[image]
`is it correct?`

yeuking, 2021-11-06:
[image]
Hope above 2 picture can help you.

1technophile, 2021-11-06:
@yeuking what is the corresponding weight and battery level when you capture these data?

yeuking, 2021-11-11:
Some Weight data capture...
0x10FFC0711F4A13880000250000000000000609596F646130

0x10FFC07406CC13880000250000000000000609596F646130

0x10FFC0770A7813880000250000000000000609596F646130

0x10FFC07B0E1013880000250000000000000609596F646130

1technophile, 2021-11-11:
Could you give the corresponding weight in kg please @yeuking

1technophile, 2021-12-07:
@yeuking could you add the value seen in the app along with each data capture please?

closed 2022-04-27 by stale[bot]