1technophile/OpenMQTTGateway #1101: Cannot compile

tablatronix, 2021-11-10:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
cannot compile for esp32devble

**To Reproduce**
/Users/shawn/projects/microcontrollers/dev/libraries/OpenMQTTGateway/main/ZgatewayBT.ino:36:11: fatal error: FreeRTOS.h:

1technophile, 2021-11-10:
Just fetched the last development branch on windows with PIO and get it built
```
Linking .pio\build\esp32dev-ble\firmware.elf
Retrieving maximum program size .pio\build\esp32dev-ble\firmware.elf
Checking size .pio\build\esp32dev-ble\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.6% (used 57808 bytes from 327680 bytes)
Flash: [=======   ]  72.4% (used 1422600 bytes from 1966080 bytes)
Building .pio\build\esp32dev-ble\firmware.bin
esptool.py v3.1
Merged 1 ELF section
========================================================= [SUCCESS] Took 87.85 seconds =========================================================
Environment    Status    Duration
-------------  --------  ------------
esp32dev-ble   SUCCESS   00:01:27.849
```

Are you trying with a fresh configuration?

tablatronix, 2021-11-10:
Could be LDF mode differences ? Not sure why pio cant find it

1technophile, 2021-11-12:
Possibly, could you share your PIO environment so that I try it?
Or are you using the default one?

tablatronix, 2021-11-12:
Yeah afaik I am using straight from the repo, ill check a few things

tablatronix, 2022-01-16:
I think I finally got this to run, no idea what the deal was, it comes and goes, seems specific to my env

closed 2022-01-16 by tablatronix

tablatronix, 2022-01-19:
I think I fixed this via
`lib_ldf_mode = chain+`