1technophile/OpenMQTTGateway #1102: Missing discovery message for BT devices

Odyno, 2021-11-12:
**Short Description**
With the current develop no discovery message are sent for BT devices

**My configuration is:**
```
platformio]
default_envs = az-delivery

[env:az-delivery]
lib_ldf_mode =  deep+
platform = espressif32
board = az-delivery-devkit-v4
check_tool = cppcheck, clangtidy
check_flags =
  cppcheck: -v
  clangtidy: -fix-errors -format-style=mozilla
#check_patterns = main/
#check_severity = medium, high

lib_deps =
  ${env:esp32dev-all-test.lib_deps}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DZgatewayRF="RF"'
  '-DLED_RECEIVE=2'
  '-DLED_RECEIVE_ON=0'
  '-DpubUnknownBLEServiceData=false'
  '-DsimpleReceiving=true'
  '-DZmqttDiscovery="HADiscovery"'
  '-DGateway_Name="OMG_BTRF_3"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="xxx"'
  '-Dwifi_password="xxx"'
  '-DMQTT_PORT="xxxx"'
  '-DMQTT_USER="xxx"'
  '-DMQTT_PASS="xxx"'
  '-DMQTT_SERVER="xxx"'
  '-DOMG_VERSION="0.9.9-1"'
  '-DLOG_LEVEL=LOG_LEVEL_VERBOSE'  ; Enable trace level logging
  '-DvalueAsASubject=true'
```

**More info**
After a code analysis, I traced back to this point and found that this variable is always = 0.

https://github.com/1technophile/OpenMQTTGateway/blob/dcbce352be38697cf66553a37b7b2a8866637aab/main/ZgatewayBT.ino#L1045

and finally I did not understand how this variable is incremented.

1technophile, 2021-11-12:
You are right I will submit a pr.

1technophile, 2021-11-12:
This is linked with the integration of [TheengsDecoder ](https://github.com/theengs/decoder), we are adding a way to retrieve the properties of a device for discovery from TheengsDecoder so that it will reduce significantly the discovery code into OMG.

1technophile, 2021-12-04:
PR done

PR #1106 (2021-12-11): Discovery with theengs decoder
## Description:
Recover the auto discovery function for BLE device through the use of [Theengs Decoder](https://github.com/theengs/decoder)
Use `std::string` for efficiency on checking devices
Close #1102 

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2021-12-11 by 1technophile