1technophile/OpenMQTTGateway #1105: actuatorONOFF no longer working with Home Assistant

josephperry, 2021-11-30:
In Home Assistant, actuatorONOFF worked in 0.9.6 but is no longer working in the 0.9.8 development version

### **ZactuatorONOFF.ino contains:**
...
#ifdef ZactuatorONOFF

#  ifdef jsonReceiving
void MQTTtoONOFF(char* topicOri, JsonObject& ONOFFdata) {
  if (cmpToMainTopic(topicOri, subjectMQTTtoONOFF)) {
    Log.trace(F("MQTTtoONOFF json data analysis" CR));
    int boolSWITCHTYPE = ONOFFdata["cmd"] | 99;
    int gpio = ONOFFdata["gpio"] | ACTUATOR_ONOFF_GPIO;
    if (boolSWITCHTYPE != 99) {
      Log.notice(F("MQTTtoONOFF boolSWITCHTYPE ok: %d" CR), boolSWITCHTYPE);
      Log.notice(F("GPIO number: %d" CR), gpio);
      pinMode(gpio, OUTPUT);
      digitalWrite(gpio, boolSWITCHTYPE);
      // we acknowledge the sending by publishing the value to an acknowledgement topic
      pub(subjectGTWONOFFtoMQTT, ONOFFdata);
    } else {
      Log.error(F("MQTTtoONOFF failed json read" CR));
    }
  }
}
#  endif
...

### **yet ZmqttDiscovery.ino contains:**
...
#  ifdef ZactuatorONOFF
  Log.trace(F("actuatorONOFFDiscovery" CR));
  char* actuatorONOFF[8] = {"switch", "actuatorONOFF", "", "", "", "{\"cmd\":1}", "{\"cmd\":0}", ""};
  //component type,name,availability topic,device class,value template,payload on, payload off, unit of measurement

  //trc(actuatorONOFF[1]);
  createDiscovery(actuatorONOFF[0],
                  subjectGTWONOFFtoMQTT, actuatorONOFF[1], (char*)getUniqueId(actuatorONOFF[1], actuatorONOFF[2]).c_str(),
                  will_Topic, actuatorONOFF[3], actuatorONOFF[4],
                  actuatorONOFF[5], actuatorONOFF[6], actuatorONOFF[7],
                  0, "", "", true, subjectMQTTtoONOFF,
                  "", "", "", "", false, // device name, device manufacturer, device model, device mac, retain
                  stateClassNone //State Class
  );
#  endif
...

Is the problem with ZmqttDiscovery.ino?

josephperry, 2021-11-30:
I should be more specific:  actuatorONOFF doesn't show as a switch in Home Assistant even with this in platformio.ini

  '-DZactuatorONOFF="ONOFF"'
  '-DACTUATOR_ON=1'
  '-DRFmqttDiscovery=true'
  '-DZmqttDiscovery="HADiscovery"'
  '-DjsonReceiving=true'

1technophile, 2021-11-30:
Thanks for the reporting this, could you check the message you have there with MQTT explorer:
[image]

josephperry, 2021-12-01:
`homeassistant/binary_sensor/30AEA474AF00connectivity/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/LWT","name":"SYS: Connectivity","uniq_id":"30AEA474AF00connectivity","dev_cla":"connectivity","pl_on":"online","pl_off":"offline","pl_avail":"online","pl_not_avail":"offline","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM","model":"[\"IR\",\"RF\",\"GPIOInput\"]"}}`

`homeassistant/binary_sensor/30AEA474AF00GPIOInput/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/GPIOInputtoMQTT","name":"GPIOInput","uniq_id":"30AEA474AF00GPIOInput","val_tpl":"{{ value_json.gpio | is_defined }}","pl_on":"HIGH","pl_off":"LOW","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM","model":"[\"IR\",\"RF\",\"GPIOInput\"]","manufacturer":"OMG_community"}}`

`homeassistant/switch/30AEA474AF00actuatorONOFF/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/LWT","name":"SYS: Erase credentials","uniq_id":"30AEA474AF00erase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_ESP32_CUSTOM/commands/MQTTtoSYS/config","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM"}}`

`homeassistant/switch/30AEA474AF00discovery/config`
`{"name":"SYS: Auto discovery","uniq_id":"30AEA474AF00discovery","retain":true,"pl_on":"{\"discovery\":true}","pl_off":"{\"discovery\":false}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_ESP32_CUSTOM/commands/MQTTtoSYS/config","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM","model":null}}`

`homeassistant/switch/30AEA474AF00erase/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/LWT","name":"SYS: Erase credentials","uniq_id":"30AEA474AF00erase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_ESP32_CUSTOM/commands/MQTTtoSYS/config","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM"}}`

`homeassistant/switch/30AEA474AF00restart/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/LWT","name":"SYS: Restart gateway","uniq_id":"30AEA474AF00restart","pl_on":"{\"cmd\":\"restart\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"home/OpenMQTTGateway_ESP32_CUSTOM/commands/MQTTtoSYS/config","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM"}}`

Auto discovery also doesn't show up as a switch in Home Assistant as it did in 0.9.6

`homeassistant/sensor/30AEA474AF00gatewayRF/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/433toMQTT","name":"gatewayRF","uniq_id":"30AEA474AF00gatewayRF","val_tpl":"{{ value_json.value | is_defined }}","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM","model":"[\"IR\",\"RF\",\"GPIOInput\"]","manufacturer":"OMG_community","sw_version":"v0.9.8+ 2021-11-30 joe"}}`

josephperry, 2021-12-01:
Oops, 30AEA474AF00actuatorONOFF wasn't copied correctly;

`homeassistant/switch/30AEA474AF00actuatorONOFF/config`
`{"stat_t":"home/OpenMQTTGateway_ESP32_CUSTOM/ONOFFtoMQTT","name":"actuatorONOFF","uniq_id":"30AEA474AF00actuatorONOFF","pl_on":"{\"cmd\":1}","pl_off":"{\"cmd\":0}","cmd_t":"home/OpenMQTTGateway_ESP32_CUSTOM/commands/MQTTtoONOFF","device":{"identifiers":["30AEA474AF00"],"name":"OpenMQTTGateway_ESP32_CUSTOM","model":null}}`

BTW using Home Assistant core-2021.11.5

1technophile, 2021-12-08:
I just tried adding this to an esp32 dev with the latest development branch
```
  '-DZactuatorONOFF="ONOFF"'
  '-DACTUATOR_ON=1'
  '-DACTUATOR_OFF=0'
  '-DACTUATOR_ONOFF_DEFAULT=ACTUATOR_ON'
```

And I got this in MQTT discovery topic:
homeassistant/switch/XXXactuatorONOFF/config

`{"stat_t":"home/XXX/ONOFFtoMQTT","name":"actuatorONOFF","uniq_id":"XXXactuatorONOFF","pl_on":"{\"cmd\":1}","pl_off":"{\"cmd\":0}","cmd_t":"home/XXX/commands/MQTTtoONOFF","device":{"identifiers":["XXX"],"name":"XXX","model":"[\"BT\"]","manufacturer":"OMG_community","sw_version":"version_tag"}}`

And I can see the switch in HASS:
[image]

josephperry, 2021-12-09:
Figured out what the real problem is: 
_when the json config response ends with **"model": null**  it is not accepted by Home Assistant._  

See my previous response for value sent for the homeassistant/switch/30AEA474AF00actuatorONOFF/config topic.  It can also effect the homeassistant/switch/30AEA474AF00discovery/config and other topics which Home Assistant then ignores.

When I reduced the length of the strings in the defines for Gateway_Name, OMG_VERSION, and the number of gateways included (which results in the "model" field) the problem goes away: actuatorONOFF and other configurations then work in v0.9.8.

Can the length of the json publishing string be increased to eliminate this problem?  I did not see this in v0.9.6, there was no **null** data.  Is there some upper bound on the permissible json length?

1technophile, 2021-12-10:
> Can the length of the json publishing string be increased to eliminate this problem? I did not see this in v0.9.6, there was no null data. Is there some upper bound on the permissible json length?

Yes it can, we are going to increase it for the new version

closed 2021-12-16 by 1technophile