1technophile/OpenMQTTGateway #1112: IBS-TH2 temperatures below around -7 degrees Celsius are incorrectly calculated

mrbubble62, 2021-12-14:
After placing the BLE IBS-TH2 in the freezer the temperature suddenly jumps to invalid value.
Current OpenMQTTGateway is reading 1,189.15 °F/643.74 °C vs. 7.93 °F/-13.38 °C for the Engbird Android application

**To Reproduce**
Place the IBS-TH2 in the freezer. When the current temperature drops below somewhere between -6 to -7 degrees Celsius the value sent to MQTT jumps +600 degrees.

**Expected behavior**
Track the value shown in the vendor application

**Environment:**
Upload from web esp32dev-ble-cont 
and clean build of esp32dev-ble from development

E.G.
```
Topic: home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/494207002890
{
  "id": "49:42:07:00:28:90",
  "name": "tps",
  "rssi": -81,
  "brand": "Inkbird",
  "model": "T Sensor",
  "model_id": "IBS-TH2",
  "tempc": 643.74,
  "tempf": 1190.732,
  "batt": 100
}
```

1technophile, 2021-12-14:
You could transform your freezer into an oven ;-)

Is it possible that the temperature is 65535 - 64374 = 1161 -> -11.61°C ? Instead of -13.38°C

1technophile, 2021-12-15:
@mrbubble62 if you are using platformio could you try to replace:
`https://github.com/theengs/decoder.git#v0.1.3`
by
`https://github.com/theengs/decoder/archive/refs/heads/negative-values.zip`
[here](https://github.com/1technophile/OpenMQTTGateway/blob/00709675bc42e09f303bb4784032ac17f248c4c5/platformio.ini#L130)

mrbubble62, 2021-12-15:
decoder = https://github.com/theengs/decoder/archive/refs/heads/negative-values.zip
This worked, thank you for such a quick response.

`Send on /BTtoMQTT/494207002890 msg {"id":"49:42:07:00:28:90","name":"tps","rssi":-92,"brand":"Inkbird","model":"T Sensor","model_id":"IBS-TH2","tempc":-10.45,"tempf":13.19,"batt":100}`

closed 2021-12-15 by mrbubble62