1technophile/OpenMQTTGateway #1123: Mi Body Composition Scale2 XMTZC05HM not working in 0.9.9beta

josephperry, 2021-12-23:
Mi XMTZC05HM Mi Body Composition Scale2 FW vers 1.0.0.12

This scale works okay with the MI Fit Android app and was recognized in OMG v0.9.6.

With v0.9.6, in MQTT Explorer (I don't know how to keep the formatting when copying from MQTT Explorer):

> homeassistant/sensor/0C9541AD7717-MiScale-weight/config
{"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"MiScale-weight",
"uniq_id":"0C9541AD7717-MiScale-weight",
"val_tpl":"{{ value_json.weight | is_defined }}",
"unit_of_meas":"kg",
"device":{
 "connections":[
   ["mac",
    "0C9541AD7717"]
   ],
  "identifiers":[
    "0C9541AD7717"
   ],
  "manufacturer":"Xiaomi",
  "model":"XMTZC0xHM",
  "name":"MiScale",
  "via_device":"OMG_ESP32_CUSTOM_BT"
  }
}

> home/OMG_ESP32_CUSTOM_BT/BTtoMQTT/0C9541AD7717
{
  "id":"0C:95:41:AD:77:17",
  "name":"MIBFS",
  "manufacturerdata":"57010c9541ad7717",
  "rssi":-71,
  "model":"XMTZC05HM",
  "weight":4.35,
  "impedance":0
}

In Home Assistant Device info shows XMTZC0xHM by Xiaomi and a MIScale-weight sensor is created as expected.

But with v0.9.9beta, in MQTT Explorer:

There is no config topic for the Mi scale

> home/OMG_ESP32_CUSTOM_BT/BTtoMQTT/0C9541AD7717
{
  "id":"0C:95:41:AD:77:17",
  "name":"MIBFS",
  "manufacturerdata":"57010c9541ad7717",
  "rssi":-76,
  "servicedata":"0384e5070c170c301d00006603"
}

Other BLE devices work okay in v0.9.9beta.

1technophile, 2021-12-26:
> Could it be that this scale identifies itself with the name MIBFS unlike other Mi scales?

We are not using the name so as to identify this scale.

Would it be possible to set [pubKnownBLEServiceData](https://github.com/1technophile/OpenMQTTGateway/blob/cf0f53507d420955ec22a07a7d07f0ee8d66594f/main/config_BT.h#L108) to `true`
 [pubBLEManufacturerData ](https://github.com/1technophile/OpenMQTTGateway/blob/cf0f53507d420955ec22a07a7d07f0ee8d66594f/main/config_BT.h#L116) to `true`
and [pubBLEServiceUUID](https://github.com/1technophile/OpenMQTTGateway/blob/cf0f53507d420955ec22a07a7d07f0ee8d66594f/main/config_BT.h#L124) to `true`

And share the MQTT message here along with what you see on the scale. This will enable me to correct the [decoder](https://github.com/theengs/decoder/blob/development/src/devices/XMTZC05HM_json.h)

DigiH, 2021-12-26:
Confirming the same issue with my Mi Body Composition Scale 2 (XMTZC05HM). The new [Theengs Decoder](https://github.com/theengs/decoder/blob/development/src/devices/XMTZC05HM_json.h) does look fine, and when setting pubBLEServiceUUID to true in config_BT.h it also shows the expected "181b" for the condition.

{
  "id":"0C:95:41:XX:XX:XX",
  "name":"MIBFS",
  "manufacturerdata":"57010c954xxxxxxxx",
  "rssi":-79,
  "servicedata":"0284e5070c1a0c2f0300001e00",
  "servicedatauuid":"0x181b"
}

Once it is clear as to why the scale isn't currently decoded as expected it might also be nice to include the previously reported impedance, for further body composition calculations, with the decoder, as posposed in the pull request

https://github.com/theengs/decoder/pull/36

DigiH, 2021-12-26:
Ok, I've changed the condition now to

`"condition":["manufacturerdata", "contain", "57010c95"],`

as this seems to be the same for @josephperry and my MiScale2 and the decoder works fine with this condition, reporting the above previously stated json, with the addition of

… ,"brand":"Xiaomi","model":"Miscale_v2","model_id":"XMTZC05HM","unit":"kg","weight":xx.x}

So the culprit seems to be the default condition

`"condition":["uuid", "contain", "181b"],`

something seems to not recognise the reported

`"servicedatauuid":"0x181b"`

although in decoder.cpp 

  const char* svc_uuid = jsondata["servicedatauuid"].as<const char*>();

servicedatauuid looks to be assigned to the conditional 'uuid', or could there be a conflict with other devices reporting a direct uuid?

DigiH, 2021-12-26:
Ok, further testing showed that now with 

`"condition":["uuid", "contain", "0x181b"],`

works as well, so the **contain** function seems gets mixed up with the hex prefix '0x'. Might possibly affect some other decoders as well.

DigiH, 2021-12-26:
Or maybe I should just relax during the holidays. Looking properly at the original decoder, there is a discrepancy between 

… contain\",\"181d\" …

and
…
"condition":["uuid", "contain", "181b"],
…

d vs. b ;)

So it's probably got nothing to do with the hex prefix. Testing to confirm ...

1technophile, 2021-12-30:
@josephperry fixed in https://github.com/1technophile/OpenMQTTGateway/pull/1124
Could you test that it works now please

josephperry, 2021-12-30:
Much better in MQTT Explorer, but Home Assistant still doesn't recognize the Mi scale weight or impedance.  However it does recognize my Xiaomi temperature & humidity sensors.

Wasn't sure how to test this:
- downloaded the latest development version
- in platform.ini only made this change: decoder = https://github.com/theengs/decoder.git#v0.1.5

MQTT Explorer homeassistant configs for the Mi scale:

> homeassistant/sensor/0C9541AD7717-XMTZC05HM-weight/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-weight",
"uniq_id":"0C9541AD7717-XMTZC05HM-weight",
"dev_cla":"weight",
"val_tpl":"{{ value_json.weight | is_defined }}",
"unit_of_meas":"kg",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
	"connections":[["mac","0C9541AD7717"]],
	"manufacturer":"Xiaomi",
	"model":"XMTZC05HM",
	"name":"Miscale_v2",
	"via_device":"OMG_ESP32_CUSTOM_BT"}
}

>homeassistant/sensor/0C9541AD7717-XMTZC05HM-impedance/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-impedance",
"uniq_id":"0C9541AD7717-XMTZC05HM-impedance",
"dev_cla":"impedance",
"val_tpl":"{{ value_json.impedance | is_defined }}",
"unit_of_meas":"Ohm",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
	"connections":[["mac","0C9541AD7717"]],
	"manufacturer":"Xiaomi",
	"model":"XMTZC05HM",
	"name":"Miscale_v2",
	"via_device":"OMG_ESP32_CUSTOM_BT"}
}

and for a Xiaomi temperature & humidity sensor (which is recognized by Home Assistant):
>homeassistant/sensor/A4C138387757-LYWSD03MMC_ATC-volt/config
{
"stat_t":"+/+/BTtoMQTT/A4C138387757",
"name":"LYWSD03MMC_ATC-volt",
"uniq_id":"A4C138387757-LYWSD03MMC_ATC-volt",
"dev_cla":"voltage",
"val_tpl":"{{ value_json.volt | is_defined }}",
"unit_of_meas":"V",
"state_class":"measurement",
"device":{"identifiers":["A4C138387757"],
	"connections":[["mac","A4C138387757"]],
	"manufacturer":"Xiaomi",
	"model":"LYWSD03MMC_ATC",
	"name":"LYWSD03MMC",
	"via_device":"OMG_ESP32_CUSTOM_BT"}
}

MQTT Explorer home data for the Mi scale:
>home/OMG_ESP32_CUSTOM_BT/BTtoMQTT/0C9541AD7717
{
"id":"0C:95:41:AD:77:17",
"name":"MIBFS",
"rssi":-77,
"brand":"Xiaomi",
"model":"Miscale_v2",
"model_id":"XMTZC05HM",
"unit":"lbs",
"weight":123.4,
"impedance":65533
}

>home/OMG_ESP32_CUSTOM_BT/BTtoMQTT/0C9541AD7717
{
"id":"0C:95:41:AD:77:17",
"name":"MIBFS",
"rssi":-71,
"brand":"Xiaomi",
"model":"Miscale_v2",
"model_id":"XMTZC05HM",
"unit":"lbs",
"weight":1.2,
"impedance":0
}

and for a Xiaomi temperature & humidity sensor (which is recognized by Home Assistant):
>home/OMG_ESP32_CUSTOM_BT/BTtoMQTT/A4C138387757
{
"id":"A4:C1:38:38:77:57",
"name":"ATC_387757",
"rssi":-82,
"brand":"Xiaomi",
"model":"LYWSD03MMC",
"model_id":"LYWSD03MMC_ATC",
"tempc":20.4,
"tempf":68.72,
"hum":49,
"batt":50,
"volt":2.675
}

For homeassistant config "unit_of_meas" is "kg", but for home "unit" is "lbs" and the weight is reported in lbs which is correct for my scale.

Anything else I should try?

josephperry, 2021-12-30:
Oops, didn't mean to close, only to comment.

DigiH, 2021-12-30:
@josephperry - what happens when you change your homeassistant configs so that the name is "MIBFS"?
…
> "connections":[["mac","0C9541AD7717"]],
> "manufacturer":"Xiaomi",
> "model":"XMTZC05HM",
> "name":"MIBFS",
> "via_device":"OMG_ESP32_CUSTOM_BT"}

josephperry, 2021-12-30:
Don't know what you mean.  Modify some of the source files and rebuild?

Could the problem be in config the unit of measure is "kg" but in home is "lbs"?

With the Xiaomi temperature & humidity sensor, temperature is reported both as tempc and tempf so Home Assistant can use either value.

DigiH, 2021-12-30:
What do you see in MQTT Explorer when stepping onto the scale?

It should definitely report

"name":"MIBFS"
and 
"model":"Miscale_v2"

at least it does here and should do so with the 0.1.5 Theengs decoder ;)

If you're still seeing 

"name":"Miscale_v2",

you might want to rebuild (clean beforehand) and upload again with platformIO.

Since I don't use Home Assistant  I'm not too sure how MQTT JSON tags are recognised there, but I meant that you might possibly need to change somthing, if previously the cale was recognised with 

"name":"Miscale_v2",

as shown in your first json message posting above. But best to start with showing your scale message in MQTT Explorer.

josephperry, 2021-12-30:
Did a clean rebuild in platformio and restarted Home Assistant.  Even reconfigured the Miscale to report in kg rather than lbs.  Home Assistant still doesn't recognize the scale.

Using MQTT Explorer to capture messages.

Same as earlier today in homeassistant config:
"manufacturer":"Xiaomi",
"model":"XMTZC05HM",
"name":"Miscale_v2",

As you expect, same as earlier today in home BTtoMQTT (except kg instead of lbs) there is:
"name":"MIBFS",
"brand":"Xiaomi",
"model":"Miscale_v2",
"model_id":"XMTZC05HM",
"unit":"kg",
"weight":17.4,
"impedance":0

As I first reported in OMG v0.9.6 (where the scale weight was recognized) homeassistant config:
"manufacturer":"Xiaomi",
"model":"XMTZC0xHM",
"name":"MiScale",

As first reported in OMG v0.9.6 home BTtoMQTT :
"name":"MIBFS",
"model":"XMTZC05HM",
"weight":4.35,
"impedance":0

Don't know what Home Assistant expects in the config message.

DigiH, 2021-12-31:
> As you expect, same as earlier today in home BTtoMQTT (except kg instead of lbs) there is:
> "name":"MIBFS",
> "brand":"Xiaomi",
> "model":"Miscale_v2",
> "model_id":"XMTZC05HM",
> "unit":"kg",
> "weight":17.4,
> "impedance":0

Yes, the correct expected json for OMG v0.9.9 beta with Theengs decoder 0.1.5.

Unfortunately my HA discovery workings knowledge is virtually non-existent.

@1technophile Any idea what the discovery issue is here?

1technophile, 2021-12-31:
I think the device class `weight `and `impedance `are not supported by HASS neither the `ohm` unit of measurement, could you try this:
```
homeassistant/sensor/0C9541AD7717-XMTZC05HM-weight/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-weight",
"uniq_id":"0C9541AD7717-XMTZC05HM-weight",
"dev_cla":"",
"val_tpl":"{{ value_json.weight | is_defined }}",
"unit_of_meas":"kg",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
"connections":[["mac","0C9541AD7717"]],
"manufacturer":"Xiaomi",
"model":"XMTZC05HM",
"name":"Miscale_v2",
"via_device":"OMG_ESP32_CUSTOM_BT"}
}

homeassistant/sensor/0C9541AD7717-XMTZC05HM-impedance/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-impedance",
"uniq_id":"0C9541AD7717-XMTZC05HM-impedance",
"dev_cla":"",
"val_tpl":"{{ value_json.impedance | is_defined }}",
"unit_of_meas":"",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
"connections":[["mac","0C9541AD7717"]],
"manufacturer":"Xiaomi",
"model":"XMTZC05HM",
"name":"Miscale_v2",
"via_device":"OMG_ESP32_CUSTOM_BT"}
}
```

josephperry, 2021-12-31:
@1technophile you're right.  Don't know why I didn't think of looking at the Home Assistant Core log before.  :-(

`2021-12-31 13:17:16 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'state_topic': '+/+/BTtoMQTT/0C9541AD7717', 'name': 'XMTZC05HM-weight', 'unique_id': '0C9541AD7717-XMTZC05HM-weight', 'device_class': 'weight', 'value_template': '{{ value_json.weight | is_defined }}', 'unit_of_measurement': 'kg', 'state_class': 'measurement', 'device': {'identifiers': ['0C9541AD7717'], 'connections': [['mac', '0C9541AD7717']], 'manufacturer': 'Xiaomi', 'model': 'XMTZC05HM', 'name': 'Miscale_v2', 'via_device': 'OMG_ESP32_CUSTOM_BT'}, 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 210, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.9/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: expected SensorDeviceClass for dictionary value @ data['device_class']`

There is a similar log message for impedance.

Since device_class is an optional [configuration variable](https://www.home-assistant.io/integrations/sensor.mqtt/) it could be removed in the weight and impedance config json.  Same for unit_of_measurement in the impedance config json.  @DigiH thanks for looking into this.

1technophile, 2022-01-03:
In this case could you publish this:
```
homeassistant/sensor/0C9541AD7717-XMTZC05HM-weight/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-weight",
"uniq_id":"0C9541AD7717-XMTZC05HM-weight",
"val_tpl":"{{ value_json.weight | is_defined }}",
"unit_of_meas":"kg",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
"connections":[["mac","0C9541AD7717"]],
"manufacturer":"Xiaomi",
"model":"XMTZC05HM",
"name":"Miscale_v2",
"via_device":"OMG_ESP32_CUSTOM_BT"}
}
```
and this:
```
homeassistant/sensor/0C9541AD7717-XMTZC05HM-impedance/config
{
"stat_t":"+/+/BTtoMQTT/0C9541AD7717",
"name":"XMTZC05HM-impedance",
"uniq_id":"0C9541AD7717-XMTZC05HM-impedance",
"val_tpl":"{{ value_json.impedance | is_defined }}",
"state_class":"measurement",
"device":{"identifiers":["0C9541AD7717"],
"connections":[["mac","0C9541AD7717"]],
"manufacturer":"Xiaomi",
"model":"XMTZC05HM",
"name":"Miscale_v2",
"via_device":"OMG_ESP32_CUSTOM_BT"}
}
```

It should work and confirm the hypothesis

josephperry, 2022-01-03:
First tried with empty strings, e.g. "dev_cla":""  and for impedance "unit_of_meas":"" and that wasn't accepted by Home Assistant.

@1technophile publishing the last messages you sent is now accepted by Home Assistant.  XMTZC05HM-weight and XMTZC05HM-impedance sensors are created for the Miscale_v2 scale.

However my Miscale_v2 scale is configured for lbs and displays and transmits in lbs not kg.  Had to change "unit_of_meas":"kg" to "unit_of_meas":"lbs" in the /0C9541AD7717-XMTZC05HM-weight/config payload to have it displayed correctly in Home Assistant.

> {
"id":"0C:95:41:AD:77:17",
"name":"MIBFS",
"rssi":-76,
"brand":"Xiaomi",
"model":"Miscale_v2",
"model_id":"XMTZC05HM",
"unit":"lbs",
"weight":123.4
,"impedance":123
}

1technophile, 2022-01-03:
Thanks for the check, I will need to think about how to handle the unity change on the discovery, maybe the tip is to don't manage the unit in this case.

josephperry, 2022-01-04:
Changed the discovery in MQTT Explore to eliminate the units for weight.  Home Assistant then displays the weight from my XMTZC05HM scale but without showing the units of measure, **lbs** in my case.  That will work.

Found this in a [fork of OpenMQTTGateway by rayslinky](https://github.com/rayslinky/OpenMQTTGateway/commit/7985927db04e8e47a0247f82fa8eef79d44e020a)

>   double weight = 0;
  if (servicedata[1] == '2') { //kg
    weight = (double)value_from_hex_data(servicedata, 22, 4, true) / 200;
    BLEdata.set("unit", "kg");
  } else if (servicedata[1] == '3') { //lbs
    weight = (double)value_from_hex_data(servicedata, 22, 4, true) / 100;
    BLEdata.set("unit", "lbs");
  } else { //unknown unit
    BLEdata.set("unit", "unknown");
  }

Could this logic be incorporated into theengs/decoder?  

Don't know for sure, this could also work for a XMTZC04HM Mi Scale configured for **lbs** rather that **kg**.

1technophile, 2022-01-10:
@josephperry could you test #1127, this PR filter measures against the list of HAAS classes available

closed 2022-01-10 by 1technophile

josephperry, 2022-01-11:
Unless I missed something I see no improvement.  Home Assistant doesn't recognize the discovery message.

Downloaded the latest development version this morning and rebuilt.

`|-- <TheengsDecoder> 0.1.5+sha.8f04985
|   |-- <ArduinoJson> 6.18.3`

Using MQTT Explorer, still seeing invalid device classes and unit of measurement in config:

> {"stat_t":"+/+/BTtoMQTT/0C9541AD7717","name":"XMTZC05HM-weight","uniq_id":"0C9541AD7717-XMTZC05HM-weight",
**"dev_cla":"weight"**,
"val_tpl":"{{ value_json.weight | is_defined }}","unit_of_meas":"kg","state_class":"measurement","device":{"identifiers":["0C9541AD7717"],"connections":[["mac","0C9541AD7717"]],"manufacturer":"Xiaomi","model":"XMTZC05HM","name":"Miscale_v2","via_device":"OMG_ESP32_CUSTOM_BT"}}

and

> {"stat_t":"+/+/BTtoMQTT/0C9541AD7717","name":"XMTZC05HM-impedance","uniq_id":"0C9541AD7717-XMTZC05HM-impedance"
**,"dev_cla":"impedance"**,
"val_tpl":"{{ value_json.impedance | is_defined }}"
**,"unit_of_meas":"Ohm"**,
"state_class":"measurement","device":{"identifiers":["0C9541AD7717"],"connections":[["mac","0C9541AD7717"]],"manufacturer":"Xiaomi","model":"XMTZC05HM","name":"Miscale_v2","via_device":"OMG_ESP32_CUSTOM_BT"}}

My XMTZC05HM scale is configured for **lbs**:

> {"id":"0C:95:41:AD:77:17","name":"MIBFS","rssi":-75,"brand":"Xiaomi","model":"Miscale_v2","model_id":"XMTZC05HM",**"unit":"lbs"**,"weight":x.x,"impedance":0}

With a 4 lb weight on the scale **configured for lbs**:

> {"id":"0C:95:41:AD:77:17","name":"MIBFS","manufacturerdata":"57010c9541ad7717","rssi":-73,
**"servicedata":"07a4e607010b0a320300008601"**,
"servicedatauuid":"0x181b","brand":"Xiaomi","model":"Miscale_v2","model_id":"XMTZC05HM","impedance":0}

and when **configured for kg**:

> {"id":"0C:95:41:AD:77:17","name":"MIBFS","manufacturerdata":"57010c9541ad7717","rssi":-74,
**"servicedata":"06a4e607010b0a372800005e01"**,
"servicedatauuid":"0x181b","brand":"Xiaomi","model":"Miscale_v2","model_id":"XMTZC05HM","impedance":0}

Should I have tried something different?  Is there another version of TheengsDecoder I should have used?

DigiH, 2022-01-11:
> Downloaded the latest development version this morning and rebuilt.

> Should I have tried something different?

Did you try with the **filter-hass-classes** branch instead if the **development** one?

https://github.com/1technophile/OpenMQTTGateway/tree/filter-hass-classes

As far as I can see that's where @1technophile submitted the change to. Hope this works for you.

josephperry, 2022-01-11:
@DigiH Thanks!  Had built with the wrong branch.  

Home Assistant now displays XMTZC05HM-impedance and XMTZC05HM-weight without units of measure. My scale is configured for lbs and the correct value is displayed.  The problem with Home Assistant XMTZ05HM discovery is corrected.

PR #1124 (2022-01-10): Bump Theengs Decoder
Fix #1123

## Description:

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
commit d34559cb (2022-01-10): Bump Theengs Decoder (#1124)

Fix #1123
PR #1127 (2022-01-12): Filter discovery messages against HASS classes list
## Description:
fix #1123 

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).