1technophile/OpenMQTTGateway #1133: Fastled MQTt exceptions

tablatronix, 2022-01-16:
Sending fastled commands crashes esp

**Describe the bug**
```php
N: [ MQTT->OMG ]: {"led":0,"hex":"330033","blink":true}
T: MQTTtoFASTLED JSON analysis
N: Led numero: 0
N: Color hex: 330033
N: Blink: 1
T: Hey I got a callback ble/OpenMQTTGateway_ESP32_BLE/commands/MQTTtoFASTLED/setbrightness
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d7a77  PS      : 0x00060530  A0      : 0x800d7b10  A1      : 0x3ffcdda0  
A2      : 0x3ffcde40  A3      : 0x00000064  A4      : 0x3ffca7fc  A5      : 0x3ffcde40  
A6      : 0x6e6f632f  A7      : 0x00000000  A8      : 0x800d7a73  A9      : 0x3ffcdd80  
A10     : 0x3ffcde40  A11     : 0x0000007b  A12     : 0x00000028  A13     : 0x3ffcde68  
A14     : 0x6e6f632f  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000070  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d7a77:0x3ffcdda0 0x400d7b0d:0x3ffcddc0 0x400dea06:0x3ffcde90 0x400deb23:0x3ffce260 0x401bf317:0x3ffce280 0x400dfff2:0x3ffce2a0 0x400dd94c:0x3ffce2e0 0x401062f0:0x3ffce300 0x40090d86:0x3ffce320

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
```
**To Reproduce**
```cpp
[env:esp32dev-ble]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
  ${libraries.decoder}
  ${libraries.fastled}  
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLED_SEND_RECEIVE=2'
  '-DLED_SEND_RECEIVE_ON=0'
  '-DZactuatorFASTLED="FASTLED"'  
  '-DGateway_Name="OpenMQTTGateway_ESP32_BLE"'
```

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
Using staging OMG
*WM: [1] ESP-IDF version:  v3.3.5-1-g85c43024c

I have not decoder trace yet

tablatronix, 2022-01-16:
hmm, a bit confusing here.

No examples for mqtttofastled syntax,

Code implies ( and i used ot use it ) simple non json commands.

But `receivingMQTT` seems to only handle json payloads

`E: deserialize MQTT data failed: IncompleteInput`

1technophile, 2022-01-18:
@BigDi do you have pointers to help @tablatronix  ?

closed 2022-04-27 by stale[bot]