1technophile/OpenMQTTGateway #1138: tempc not unsigned

tablatronix, 2022-01-18:
Latest, my Mi sensors do not report -C temps properly.

`BLEdata["tempc"] = (float)((pData[0] | (pData[1] << 8)) * 0.01);`

tablatronix, 2022-01-18:
"tempc":648.16 ?
{"id":"A4:C1:38:xx:xx:xx","name":"ATC_xxxxxx","rssi":-83,"model":"LYWSD03MMC_PVVX","tempc":648.16,"tempf":1198.688,"hum":74.52,"batt":45,"volt":2.613}

1technophile, 2022-01-18:
This should be corrected in v0.9.9beta, are you using a previous version?

tablatronix, 2022-01-18:
I should be using staging.. how do I check ?

`N: OpenMQTTGateway Version: version_tag`

1technophile, 2022-01-18:
When did you took this branch?

By the way a simple test is to do a web upload of the v0.9.9beta:
https://docs.openmqttgateway.com/upload/web-install.html

tablatronix, 2022-01-18:
Yesterday..

9a4db88cb251bd4ec21ccda6acfe2890de8983c9

1technophile, 2022-01-18:
Hmm, could you set [this ](https://github.com/1technophile/OpenMQTTGateway/blob/8b40b680811753c6cd80097657e9e82dda6aa33c/main/config_BT.h#L108) to true and share the servicedata here please

tablatronix, 2022-01-18:
Of the sensor? Let me put it in the fridge

1technophile, 2022-01-18:
Yes please

1technophile, 2022-01-18:
Are you using platformio or arduino ide?

With arduino ide you may have to update to the[ latest version ](https://github.com/theengs/decoder/releases)of Theengs Decoder

tablatronix, 2022-01-19:
Its fixed, I must have used an older bin when my ide was having issues, I was looking at historical data.
Sorry about that

closed 2022-01-19 by tablatronix

1technophile, 2022-01-19:
Thanks, i wilk use these data to enrich the device decoding unit testing