1technophile/OpenMQTTGateway #1139: Ibeacons with random MAC addresses

scribblrsam, 2022-01-18:
**Is your feature request related to a problem? Please describe.**

I would like to use the home assistant companion app to publish beacons over ble, collect these and track presence of people in the house like this. As far as I can see this is not currently possible as modern phones randomise the Bluetooth MAC address used and presence events are published with the mac as the device ID.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Publish ibeacon presence events to HA/MQTT using the beacon uuid not the MAC or perhaps both to allow either event to be tracked. I am happy to look at this, would you consider a PR for this?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Carry a hardware ibeacon, clearly less practical and I don't always have the same set of keys!

I have also considered using esphome as this has functionally to track beacons by uuid, I would rather use OMG as I use this for other sensors. 

**Additional context**
Add any other context or screenshots about the feature request here.

1technophile, 2022-01-18:
>Publish ibeacon presence events to HA/MQTT using the beacon uuid not the MAC or perhaps both to allow either event to be tracked. I am happy to look at this, would you consider a PR for this?

You should [get the UUID](https://github.com/theengs/decoder/blob/3abed86931ffec8890ced8a3b12a05a55f92bd05/src/devices/iBeacon_json.h#L13) already with an ibeacon

With the v0.9.9 beta

Which version are you using?

scribblrsam, 2022-01-19:
> Publish ibeacon presence events to HA/MQTT using the beacon uuid not the MAC or perhaps both to allow either event to be tracked. I am happy to look at this, would you consider a PR for this?

Sorry this was meant to say: 
Publish ibeacon presence events to HA/MQTT using the beacon uuid * as the device ID and* not the MAC or perhaps both to allow either event to be tracked. I am happy to look at this, would you consider a PR for this?

The messages are published to the `home/home_presence/<room name>` topics
`{"id":"50:54:99:B4:00:18","rssi":-82,"distance":11.4949,"brand":"GENERIC","model":"IBEACON","model_id":"IBEACON","mfid":"4c00","uuid":"e1e4448b54ea4a7eafc833522f901b6c","major":100,"minor":1,"power":-59}`

The mac addresses used for the ID are ever changing and therefore cannot be used to track a device according to [the HA docs for MQTT room presence](https://www.home-assistant.io/integrations/mqtt_room/). Using the uuid as the ID would allow these beacons to be tracked. 

[ESP Home ](https://esphome.io/components/binary_sensor/ble_presence.html) allows for similar tracking of UUID rather than MAC for this reason however only creates a binary sensor.

scribblrsam, 2022-01-19:
Is that not the macro that publishes the message above to the home_presence topic?

1technophile, 2022-01-19:
> Is that not the macro that publishes the message above to the home_presence topic?

Yes

scribblrsam, 2022-01-23:
Unless I'm missing something it doesn't work for my case in its current state. The mac of the device is published in the Id field. This is used in the home assistant config to track a device.

As beacons are being published from a phone the mac is randomised the beacons cannot be tracked. If the beacon uuid was to be used as the id a beacon from a phone could be tracked.

Although this would also fail to pass whitelist checks...

Can you recommend a solution?

h2zero, 2022-01-23:
I think this would be doable. A simple macro to set `useBeaconUUIDasID`, then check if it's an iBeacon before publishing and change the ID to the uuid.

h2zero, 2022-01-23:
@scribblrsam Here is a simple patch for you to try. Let me know if this is sufficient.
[uuid_as_id.txt](https://github.com/1technophile/OpenMQTTGateway/files/7920727/uuid_as_id.txt)

ssikgun, 2022-01-28:
Hello. I have the same problem. How did you solve it?? The Bluetooth MAC address of my phone also changes randomly. So I want to use the UUID as the device ID.

h2zero, 2022-01-28:
@ssikgun Try the patch I provided in my post above.

scribblrsam, 2022-01-28:
@h2zero thanks for the patch. I tired it but couldn't get it to work in conjunction with the mac whitelist. 

I did start to look at this further with logging set to trace but didn't get very far as I'm not familiar with the project and haven't had much time to spend on this.

Is there anything you can suggest?

h2zero, 2022-01-28:
There would need to be much more code alteration for whitelist support unfortunately. Not a 5 minute change like the patch. 

If you want to hack around checkout the processBLEtask function.

ssikgun, 2022-01-28:
@h2zero sorry. I don't know how to apply the patch. So, I directly modified the code using platformio.
(Fixed ZgatewayBT.ino)
[image]

And set the mqtt-room in ha.

[image]

But it didn't work.
[image]

What did I do wrong? 

And thanks for the reply.

h2zero, 2022-01-29:
Did you add `-DuseBeaconUuidAsId` to your env config? Alternatively just remove the `#if` and `#endif` parts of the patch.

ssikgun, 2022-01-29:
@h2zero Thank you very much. I modified the ZgatewayBT code.
[image]

It finally started to work.
[image]

I will share it with people who have the same problem as me on the home assistant community in South Korea.

mihsu81, 2022-03-25:
Hi @1technophile, Can this be included in the next release?
Looks like a good solution.

1technophile, 2022-03-25:
@mihsu81 yes indeed

mihsu81, 2022-03-25:
Awesome. Having the UID as device id will allow the UID to be included in the whitelist?

1technophile, 2022-03-26:
That's a good question, I will need to identify the impacts of doing so.

closed 2023-10-20 by github-actions[bot]