1technophile/OpenMQTTGateway #1141: Deactivate jsonPublishing from `build_flags` in  `*_env.ini`

debuglevel, 2022-01-20:
**Describe the bug**
If I see it correctly `jsonPublishing` cannot be deactivated without actually editing `main/User_config.h`:

If you would try to undefine it via 
```ini
build_flags =
  ${env:esp32-lolin32lite-ble.build_flags}
  '-UjsonPublishing'
```
it will be set again in `main/User_config.h`:

```c
#ifndef jsonPublishing
#  define jsonPublishing true //comment if you don't want to use Json  publishing  (one topic for all the parameters)
#endif
```` 

Setting
```ini
build_flags =
  ${env:esp32-lolin32lite-ble.build_flags}
  '-DjsonPublishing=false'
```
probably also has no effect, as the code in `main.ino` just checks if it is defined, but not for its value:
```c
#ifdef jsonPublishing
    Log.trace(F("jsonPublishing" CR));
#  if defined(ESP8266) || defined(ESP32) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)
    char JSONmessageBuffer[data.measureLength() + 1];
#  else
    char JSONmessageBuffer[JSON_MSG_BUFFER];
#  endif
    data.printTo(JSONmessageBuffer, sizeof(JSONmessageBuffer));
    pubMQTT(topic, JSONmessageBuffer);
#endif
```

Unless I'm mistaken (sorry then!), one cannot deactivate `jsonPublishing` from in  `*_env.ini` (btw: which is great for the other configurations!)

I think `#if jsonPublishing == false` in `main.ino` could do the trick. But I'm a lousy C programmer and would rather leave the solution up to someone who know's what he's doing :)

**Environment (please complete the following information):**
* tag `v0.9.8` from git
* compiled via `platformio`

1technophile, 2022-01-20:
Good catch,
what is the use case for deactivating it? For my curiosity.

I'm planning from a long time to remove the `simplePublishing` and to have permanently `jsonPublishing`

debuglevel, 2022-01-21:
To be honest, I just wanted to start with the "plain MQTT" thing for now - and I like the "hierarchical topic approach" of MQTT, even though JSON is probably also great for backend processing.

I think it would be nice to keep both approaches. There are probably some processing backends which can handle the "traditional" MQTT way but have no idea what to do with a JSON string. If other small IoT devices want to subscribe to the JSON topic and have to deserialize it, I could imagine that they could possibly run into memory issues. But I'm quite new to this whole IoT thing - no string opinion here. But having the choice is always nice in a relatively new and evolving field like IoT. Just some random thoughts as a software developer/engineer/whatever :)

My next steps in trying out things would probably to write a subscribing client which just throws some topics/sensor data into a time-series database (like InfluxDB). If the data would be a JSON string, I would have to subscribe the topic, deserialize JSON and access the field. If it was "just a topic with a plain value" I would just access the value. Although deserizalizing JSON is not a big thing, the later feels to be the more elegant way (for me at this time) - but I clearly see the value of JSON which can at least differentiate between some basic data types. :-)

closed 2022-04-27 by stale[bot]