1technophile/OpenMQTTGateway #1149: No reset of WIFI credentials by TRIGGER_GPIO with ESP8266. AP mode never entered again.

nerk, 2022-01-26:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**

My gateway uses an ESP8266.

Once a WIFI connection has been successful just once, there is no way to put the gateway back into AP mode without explicitly erasing the flash prior to flashing. Defining TRIGGER_GPIO and using the button only erases SPIFFS and clears the MQTT settings, but does not clear the WIFI connection information.

**To Reproduce**
Steps to reproduce the behavior:
1. Use ESP8266, define TRIGGER_GPIO, use WifiManager, compile and flash
2. Connect serial monitor
2. Enter correct WIFI credentials and MQTT configuration and connect to WIFI
3. Press the trigger button for more than 3 seconds

**Expected behavior**
I expect that the gateway clears all configuration information and restarts in AP mode. However, only the MQTT configuration is cleared, the gateway connects to the existing WIFI network and tries to communicate with mqtt broker 192.168.1.17, which is the default.

**Environment (please complete the following information):**
* OpenMQTTGateway version used: development
* ESP8266

**Additional context**

The following code seems to be the problem
```
void eraseAndRestart() {
  Log.trace(F("Formatting requested, result: %d" CR), SPIFFS.format());

#  if defined(ESP8266)
  SPIFFS.format();
  delay(5000);
  ESP.reset();
#  else
  nvs_flash_erase();
  ESP.restart();
#  endif
}
```
In case of an ESP8266, only SPIFFS is cleared, not flash. `nvs_flash_erase()`is only invoked for ESP32.
 
Could we just use `wifiManager.erase()` in both cases or is there a particular reason for `nvs_flash_erase`? It also seems that `SPIFFS.format()`is called twice for an ESP8266, first in the Log.trace statement, second in the ESP8266 section.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

nerk, 2022-01-26:
```
void eraseAndRestart() {
  Log.trace(F("Formatting requested, result: %d" CR), SPIFFS.format());
  wifiManager.erase();
  delay(5000);
  wifiManager.reboot();
}
```

This seems to work fine (not tested with ESP32). `delay` is only to make sure to see messages on serial monitor.

1technophile, 2022-01-26:
@nerk could you try the linked PR?

nerk, 2022-01-27:
@1technophile This PR fixes the problem.

PR #1150 (2022-01-27): ESP8266: Fix partial erase of connection settings
Fix #1149

## Description:

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2022-01-27 by 1technophile