1technophile/OpenMQTTGateway #1155: Unit of measurement in OMG 0.9.9 sensor

mr-orange, 2022-02-04:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Whit the latest OMG version some of the device sensor have lost the unit of measurement.
As an example you can see SYS: Free memory or SYS: Uptime and others. Also happen on some of miflora sensor like fertility and moisture. 
The version 0.9.8 does not have this bug.

**To Reproduce**
Steps to reproduce the behavior:
1 Always happening

**Expected behavior**
A clear and concise description of what you expected to happen.
All the sensor need to have the proper unit of measurement attached in mqtt discovery data

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 OpenMQTTGateway version used V0.9.9
 Library version related to the problem you have BLE and RF

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-02-05:
I see from where it is coming from, I suppose that you are using home assistant?
It seems that the filtering introduced with #1127 need to be separated for the units.
I will push a PR

1technophile, 2022-02-05:
PR published, could you try it out please?

mr-orange, 2022-02-06:
Hi Florian thanks for the fix.  I can try out the fix only on Monday. And yes I use homeassistant. Need I compile the latest development branch? Or need I  upload the binary?
Anyway I can’t see the fix on omg GitHub repository.
Let me know.

DigiH, 2022-02-06:
> Anyway I can’t see the fix on omg GitHub repository.

@mr-orange Florian submitted the fix to the **unit-filter** branch

https://github.com/1technophile/OpenMQTTGateway/tree/unit-filter

PR #1159 (2022-02-08): HASS auto discovery - Separate units filtering from classes
## Description:
fix #1155 

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2022-02-08 by 1technophile