1technophile/OpenMQTTGateway #1156: v0.99 does not recognize LYWSD03MMC PVVX devices anymore

kgarrels, 2022-02-05:
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
v0.99 does not recognize LYWSD03MMC PVVX devices anymore, I only receive raw "servicedata" messages.
V0.98 recognizes these without problem.

**To Reproduce**
install V0.99 - not recognized
install v0.98 - recognized

**Expected behavior**
changing from V0.98 to V0.90 should not change behaviour

**Environment (please complete the following information):**
* OpenMQTTGateway V0.99
* m5stack

DigiH, 2022-02-05:
This has been fixed in the theengs decoder

https://github.com/theengs/decoder/pull/65

Updating to the latest version of dev or #ef7c565 will fix it for you.

kgarrels, 2022-02-05:
perfect, thanks a lot!

closed 2022-02-05 by 1technophile