1technophile/OpenMQTTGateway #1161: MQTT reconnection error: OpenMQTTGateway already connected, closing old connection.

mr-orange, 2022-02-07:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
My 0.9.9 cc1101 - RF gateway disconnect and reconnect to mqtt client  with error: err mosquitto[14343]: Client OpenMQTTGateway_multi_receiver already connected, closing old connection. 
This happen every minutes and in some cases every 10 or 30 seconds.
If the error is reported for long period of times the lan connection to the device is completely lost.
On the same lan the Olimex WIFI BT gateway does present this error sporadically only 2 or 3 per day.
Let me know how I can help you to find this bug.
 
**To Reproduce**
Wait a few minutes after restart ?

**Expected behavior**
No reconnection error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-02-08:
Could you check that the gateways have 2 different names, frequent disconnections can be due to having several gateways configured with the same client_id (=gateway_name)

mr-orange, 2022-02-08:
I see different names in mqtt explorer, but i do not know if the name is
the client id.
[image: immagine.png]

in my system log I have:
[image: immagine.png]

In Home Assistant I see:
[image: immagine.png]

1technophile, 2022-02-14:
I cannot access the images unfortunately

mr-orange, 2022-02-18:
Hi Florian I've switched my cc1101 device to pilight version and the gateway is rocksolid on my lan no more connection error on my syslog and gateway reboot my uptime know is around 8 days.
I think the problem relies non the multireceiver version  and the problem can be memory related. In pilight only version I actually have 240K of free memory while on multi receiver I had around 100K.
Probably there is a problem that lead to reboot the esp32 so the device don't know anything about the last connection and during the connection after reboot the syslog error arise.
Let me know if I can hel you to track down the problem I will reflash my device.

NorthernMan54, 2022-02-18:
@mr-orange can you share the details needed for recreating the issue?  Ie which build, any custom directives applied and any other relevant details.

mr-orange, 2022-02-22:
Sorry for the late answer. The issue decribed above was with the 0.9.9 multireceiver version.
I use the gateway with home assistant and have a script that fire a loop of pylight rf code every  250ms until the outlet is on or off depending on the code.
I think but do not have any proof the gateway reboot and do a reconnection that it's not expired on the broker and so the error arise. 
Viewing I saw a lot of mqtt log in syslog I see a lot of OpenMQTTGateway already connected, closing old connection.
Let me know how I can help.

closed 2023-10-20 by github-actions[bot]