1technophile/OpenMQTTGateway #1163: Is it possible to get all the separate manufacturerdata?

saramlove, 2022-02-10:
**Is your feature request related to a problem? 
I want to get the second manufacturerdata of the Apple device.
Currently only the first of all manufacturerdata can be obtained.

**Describe the solution you'd like** A clear and concise description of what you want to happen.
I can get the info I want in the second manufacturerdata.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
I want to get multiple like servicedata .

**Additional context**
Add any other context or screenshots about the feature request here.

Hello.
Get only the first of several manufacturerdata.
Is it possible to collect all the manufacturerdata?

OMG is the best BLE data analysis solution.
Thank you.
[image]

h2zero, 2022-02-11:
This is interesting, I've not seen a device with multiple manufacturing data definitions before. This will require updating of the NimBLE repo to support correctly and could take some time.

saramlove, 2022-02-11:
> This is interesting, I've not seen a device with multiple manufacturing data definitions before. This will require updating of the NimBLE repo to support correctly and could take some time.

Thank you very much for your reply.  : )

closed 2022-04-27 by stale[bot]