1technophile/OpenMQTTGateway #1165: BLE READ/WRITE with new "immediate" parameter causes restart when BLE device is not available

DigiH, 2022-02-11:
While changing over my openHAB backend code to use the new "immediate": true parameter for my READ and WRITE BLE commands I found that it causes the BLE OMG gateway to reset if the relevant BLE device is not available/switched off, while the same command, without the "immediate" parameter, or "immediate": false, eventually comes back with a "success":false reply.

While a very low ttl value of 1 or 2 usually doesn't cause this, any higher ttl, especially with 5+ consistently reproduce this reset. I assume that is has something to do with the number of connection attempts and my related setting of

'-DTimeBtwRead=20000'

This constant reset for unreachable devices causes them to never reply with "success":false, therefor never indicating that they might be offline or unreachable.

Tested with latest OMG 0.9.10 and the same commands used in the initial "immediate" parameter testing

https://github.com/1technophile/OpenMQTTGateway/pull/1137

e.g.

`{"ble_read_address":"AA:BB:CC:DD:EE:FF","ble_read_service":"180f","ble_read_char":"2a19","value_type":"HEX","ttl":10,"immediate":true}`

DigiH, 2022-02-12:
> It's better to handle the "success:false" message to retry connection for this purpose instead of TTL at this time. I will consider how to improve this in a future update.

Thanks @h2zero, In further testing I found even with a ttl value of 2 the reset occurs quiet regularly.

Am I right in assuming that this could be cause when the delay >10 seconds falls into the next scheduled scan that this happens?

Setting ttl to 1 doesn't cause resets, but also doesn't always connect on the 1st and only try ;)

I would be great if this can be improved in a future update, to avoid a reset and eventually give a "success":false reply, similar to now with higher ttl values when the **immediate** parameter is false.

Could this issue be changed to a future enhancement please?

h2zero, 2022-02-12:
Can you provide a log of the reset? I'd like to see what error is when it occurs.

DigiH, 2022-02-12:
I've been thinking about that already, but need to dig out my BLE gateway, which is tucked away very awkwardly and only has been getting OTAs for a long time ;)

Spare ESP32s are also already on the way though. Will provide a log later today, or tomorrow at the latest.

DigiH, 2022-02-12:
```
: [ MQTT->OMG ]: {"ble_read_address":"AA:BB:CC:DD:EE:FF","ble_read_service":"180f","ble_read_char":"2a19","value_type":"HEX","ttl":2,"immediate":true}
N: BLE Connect begin
E NimBLEClient: "Connection failed; status=13 "
E: Connect to: aa:bb:cc:dd:ee:ff failed
E NimBLEClient: "Connection failed; status=13 "
E: Connect to: aa:bb:cc:dd:ee:ff failed
N: BLE Connect end
N: BLE Connect begin
N: BLE Connect end
N: Send on /BTtoMQTT/ACE64B07CD95 msg {"id":"AA:BB:CC:DD:EE:FF","service":"0x180f","characteristic":"0x2a19","success":false}
W: Client not connected, aborting thes publication
W: MQTT connection...
N: Connected to broker
```

while the last 

`{"id":"AA:BB:CC:DD:EE:FF","service":"0x180f","characteristic":"0x2a19","success":false}`

is not being sent to the broker, with
`
W: Client not connected, aborting thes publication`

and causing the reset.

Shall I do more logging with _TRACE, _VERBOSE ... ?

DigiH, 2022-02-12:
> Oh so it's not crashing the device, but just dropping the MQTT connection?

Hmm, now that I can do the monitoring ;) Previously I assumed it was resetting the gateway, as the broker showed the LWT offline/LWT online routine ;)

My main problem being that that last 

```
N: Send on /BTtoMQTT/AABBCCDDEEFF msg {"id":"aa:bb:cc:dd:ee:ff","service":"0x180f","characteristic":"0x2a19","success":false}
T: jsonPubl - ON
W: Client not connected, aborting thes publication
```

"success":false reply is never being sent to the broker.

Any reasson why this is happening with a tll of just 2, when with my previously used ttl=10 I never had these connection resets, even if after 10 tries the reply was false when all devices were ON?

h2zero, 2022-02-12:
>Any reasson why this is happening with a tll of just 2, when with my previously used ttl=10 I never had these connection resets, even if after 10 tries the reply was false when all devices were ON?

The call to connect immediately is happening within the MQTT command handler, which now blocks and waits for the result before returning.
The TTL value will loop on connection failure within this call from the MQTT command handler and if this takes too long before returning the keep alive timer will expire and disconnect the client. This is different from the older versions due to the "immediate" functionality added, which necessitated this method. One simple workaround for now would be to increase the keep alive time.

DigiH, 2022-02-12:
Thanks for the explanations @h2zero. I've currently reverted to checking the battery status without the **immediate** parameter, as it's not really necessary for this reading, as I've done previously and increased the ttl back to 10 again. This gives me a pretty reliable status if any of the devices might be switched off, therefor not using any WRITE commands with the **immediate** parameter.

h2zero, 2022-02-13:
In the future this should probably be handled in it's own task. I will put some thought into resolving this before the next release.

closed 2022-04-27 by stale[bot]

h2zero, 2022-04-28:
@DigiH The switchbot PR has been updated and adds a task for all BLE commands that should eliminate this issue.

DigiH, 2022-04-28:
@h2zero Thanks a lot, just tested it and working great with my Playbulb Candle battery check routine, even for the ones which are turned off/unreachable.