1technophile/OpenMQTTGateway #1166: Mi Scale not decoded

kgarrels, 2022-02-12:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
the Miscale without impedance is not detected.

**To Reproduce**
```
{"id":"70:87:9E:5B:33:42","mac_type":0,"manufacturerdata":"570170879e5b3342","rssi":-79,"servicedata":"a2e45ce607020c001103"}
```

**Expected behavior**
Decoding of correct weight

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.9 development)

kgarrels, 2022-02-12:
from the serial debug:

```
I NimBLEScan: "Updated advertiser: 70:87:9e:5b:33:42"
T: Creating BLE buffer
N: Device detected: 70:87:9E:5B:33:42
T: getDeviceByMac 70:87:9E:5B:33:42
T: Get services data number: 1
T: Converted service data (10) to a2f45be607020c092d23
T: Service data: a2f45be607020c092d23
T: Service data UUID: 0x181d
T: Removing servicedatauuid
N: Send on /BTtoMQTT/70879E5B3342 msg {"id":"70:87:9E:5B:33:42","mac_type":0,"name":"MI SCALE2","manufacturerdata":"570170879e5b3342","rssi":-78,"servicedata":"a2f45be607020c092d23"}
```

kgarrels, 2022-02-12:
the printing on scale says "Mi Smart Scale 2 Model XMTZC04HM".
Date code on the product "2021.02".

It is without impedance sensor.

DigiH, 2022-02-12:
> the printing on scale says "Mi Smart Scale 2 Model XMTZC04HM".

Ok, so it's definitely a XMTZC04HM, without the impedance sensor, and the current decoder condition does not recognise the "2021.02" **"servicedata", "index", 0, "a"** condition.

All other decoding might still be the same though. Is it correct that you are using the **kg** unit on your scale and not **lbs**?

Are you building your own binaries to upload the OMG gateway, and would you be able to test an amended decoder?

kgarrels, 2022-02-12:
yes, I have managed to build from source.

Nice project BTW, thanks for your efforts!

thanks,
kai

DigiH, 2022-02-12:
> Nice project BTW, thanks for your efforts!

That definitely goes to @1technophile @h2zero @NorthernMan54 and many other regular hard core contributors! I'm just a user who tries to chip in when I feel I can be of help :)

I've made a quick test fork, so if you can change the decoder library to

```
decoder = https://github.com/DigiH/decoder
```

in platform.ini, to see if your scale is now recognised, and also if the reported weight values are in sync with what is showing on the scale.

kgarrels, 2022-02-12:
that is so :-(

kgarrels, 2022-02-13:
it works, but now we are also sending "unstabilzed" weights, that's not good.
The stabilzed bit is bit 10 in the control world at the beginning, see here:
https://github.com/wiecosystem/Bluetooth/blob/master/doc/devices/huami.health.scale2.md

This was changed back and forth in https://github.com/theengs/decoder/pull/53

Is there a description of the "condition laguage" somewhere?

kgarrels, 2022-02-13:
continue here:
https://github.com/theengs/decoder/pull/67

kgarrels, 2022-02-13:
correct, but it is index 0 (not 2).
I think the need to have the "weight removed" bit false was intentional, maybe to have the time correct.
I suggest to leave it like so.

Thanks again for your help!

closed 2022-02-13 by kgarrels

kgarrels, 2022-02-14:
For me, 'a' is the better and more likely state.

In think there was a reason behind using '2', somewhere I read about the timestamp in the service data being only correct then (make sense...). As I send the value to influxdb, I do not care about the time, I gets timestamped during ingestion.

So I think that the condition should be written as "either '2' or 'a'", but I don't know how to write that in the JSON language.

DigiH, 2022-02-14:
@kgarrels I agree, **'a'** is far more robust in reliably registering the last weight.

Form my testing with the XMTZC05HM there is no other change in the service data - also not the time stamp - apart from the **'2'** to **'a'** switch. Can you also confirm that for the XMTZC04HM.

Since registering both options, which I don't really think necessary, when a single switch to **'a'** should be sufficient it would need  a nested OR within the AND condition already existing. Something like

`"condition":["uuid", "contain", "181d", "&",("servicedata", "index", 0, "2", "|", "servicedata", "index", 0, "a")],`

which I don't think is possible in the decoder yet. I'm already discussing this in the decoder discussions.

But if you can also confirm the above identical time stamp for the **'2'** and **'a'** messages it would make a good case to shift the **stabilised** flag for both scale versions to **'a'**.

Thanks