1technophile/OpenMQTTGateway #1168: Allow whitelisting/blacklisting of device types

kabili207, 2022-02-16:
**Is your feature request related to a problem? Please describe.**
One of my gateways is near an iBeacon which seems to be using random MAC addresses and UUIDs, which is causing a flood of devices to show up in home assistant. I'm talking literally hundreds of devices.

**Describe the solution you'd like**
I would love to see whitelisting and blacklisting based on the device type in addition to the existing MAC based filtering.

**Describe alternatives you've considered**
I've considered using MAC whitelisting, but I mess with my gateways and sensors fairly often and it's very annoying to need to constantly update the filter list.

**Additional context**
Add any other context or screenshots about the feature request here.

1technophile, 2022-02-26:
I understand, I will think of a solution to avoid this

psarossy, 2022-04-16:
I've ran into this as well, windows machines, ibeacons, TPMS sensors and now some thing with "brand":"GENERIC","model":"GAEN","model_id":"GAEN" is showing up every few seconds and creates a new entry... It create 32k entities before I realized...

Could the black-list be configurable for model/brand etc as well, not just mac address?

1technophile, 2022-04-16:
> I've ran into this as well, windows machines, ibeacons, TPMS sensors and now some thing with "brand":"GENERIC","model":"GAEN","model_id":"GAEN" is showing up every few seconds and creates a new entry... It create 32k entities before I realized...
Could the black-list be configurable for model/brand etc as well, not just mac address?

I'm going to publish a new version that filters these devices from discovery.

eloo, 2022-05-02:
Hi, 
i'm also facing this issue and its getting out of hands :D 
[image]

Can't wait to disable this devices ;)

1technophile, 2022-05-07:
Hi,

The new version v0.9.12 doesn't discover GAEN, MS_CDP and IBeacon, you should not have this issue anymore.

eloo, 2022-05-10:
@1technophile can confirm that i don't see any new devices of GAEN MS_CDP or IBeacon anymore.
Thanks for this fix 👍

closed 2022-05-10 by 1technophile