1technophile/OpenMQTTGateway #1175: deduplication for ZgatewayRTL_433/

billy27607, 2022-03-02:
Would it be possible to add message deduplication to ZgatewayRTL_433? A number of the sensors, Acurite for example, send 3 copies of each message. It would great to filter them out.

NorthernMan54, 2022-03-03:
I personally have a fair number of acurite sensors, and have not noticed this to be a issue to address.  I have tried to keep the library as close as possible to the stock rtl_433 code base to make updating and keeping current easy.

My clients node-red and homebridge don’t really care about any duplicates, they just process the update.  And for any graphs I do with the data I added a data normalization layer as all sensors have different rates of updates.

billy27607, 2022-03-03:
I guess that how one handles data and expectations of how clean it needs to be is a matter of preference. I have been using the sensors for sometime using a Pi as a gateway with rtl_433 and a python script that I wrote to deduplicate, clean and publish to mqtt. Some of the sensors that I have do not send messages in triplicate and some do and the message count is data as well. 

deduplicating is not an absolute requirement, but it would keep the rtl_433 consistent with the rf, ir, and weather station. When I get around to it, I'll see if I can code it up so that it is consistent with the OMG coding style and enter a pull request.

closed 2023-10-19 by github-actions[bot]