1technophile/OpenMQTTGateway #1176: Feature Request: Support for SwitchBot

alfista2600, 2022-03-04:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Support for SwitchBot mechanical button presser
https://www.switch-bot.com/products/switchbot-bot

**Describe alternatives you've considered**
Not automating the device in question

**Additional context**
I have a device and am happy to help however I can. Not really a developer, but a lifetime in technology.

DigiH, 2022-03-04:
Hi @alfista2600, have a look at this work in progress pull request

https://github.com/1technophile/OpenMQTTGateway/pull/995

alfista2600, 2022-03-04:
closed per @DigiH's link. I need to remember to look for feature requests in the future. Now to try to figure out how to add this code to my build while I wait for it to be formally included!

closed 2022-03-04 by alfista2600

1technophile, 2022-03-04:
You can already trigger a switchbot without modifying the code, just replace the MAC Address below with the one of your switchbot:
`home/OpenMQTTGateway/commands/MQTTtoBT/config`

```json
{
  "ble_write_address": "FF:AA:BB:FF:DD:EE",
  "ble_write_service": "cba20d00-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_char": "cba20002-224d-11e6-9fb8-0002a5d5c51b",
  "ble_write_value": "570100",
  "value_type": "HEX",
  "ttl": 4,
  "mac_type":1,
  "immediate": true
}
```

alfista2600, 2022-03-04:
Neat! Thanks! Got it working. I've gotta figure our more of this stuff!