1technophile/OpenMQTTGateway #1179: Public BSSID as a SYS sensor

chinyaolin, 2022-03-07:
I have a card in Homeassistant to show how my wifi end devices connect to.
I use BSSID to determine whitch "AP" they connected.
IT would be nice to get OMG BSSID.
[image]

1technophile, 2022-03-10:
Hello,

Seems that a method exists for this `WiFi.BSSID()`, would you like to submit a PR to add it after [here](https://github.com/1technophile/OpenMQTTGateway/blob/a1846faee9d54f8e64fe97096f08bad37ff350ee/main/main.ino#L1558)?
```cpp
  String BSSID = WiFi.BSSID();
  SYSdata["BSSID"] = BSSID;
```

chinyaolin, 2022-03-30:
Its my first time to create a PR.
Let me know if anything wrong.

closed 2022-03-31 by chinyaolin