1technophile/OpenMQTTGateway #1189: Restart causes loss of config/Whitelist not working

anarchking, 2022-03-19:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Publish to home/Bedroom_OMG_BLE/commands/MQTTtoBT/config
Payload {"white-list":["xx:xx:xx:xx:xx:xx","xx:xx:xx:xx:xx:xx"]}

3. Subscribe on home/home_presence/Bedroom_OMG_BLE
4. See error
I updated from OMG 9.8 to 9.11 and cant seem to get it back to the way i had it. 

**Expected behavior**
A clear and concise description of what you expected to happen. When first starting the device after setting up Wi-Fi and mqtt creds, if I switch on ha presence I'm flooded with all kinds of eye beacon and other Bluetooth devices automatically discovered if I try white listing nothing is discovered if I restart the device it loses all config not sure what I'm doing wrong.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-03-19:
>  if I switch on ha presence I'm flooded with all kinds of eye beacon and other Bluetooth devices automatically discovered if I try white listing nothing is discovered if I restart the device it loses all config not sure what I'm doing wrong.

This is corrected on the development branch, we are filtering out those devices.

Did you tried:
` {"white-list":["xx:xx:xx:xx:xx:xx","xx:xx:xx:xx:xx:xx"],"interval":3000}`

anarchking, 2022-03-19:
Awesome 👍, thanks I figured out my problem was that my HA wasn't following the state of the (publish Hass presence switch), so it looked like it was on but when I clicked the button it fixed the problem. For now setting config on restarts I'm just using an automation to send all config to each device. I didn't even think about adding the payloads together thanks for that tip as well 😁

closed 2022-03-19 by anarchking