1technophile/OpenMQTTGateway #1191: configuration blocks the work nodemcuv2-rf-cc1101.bin  GPIO 0 (D3)

Monya2016, 2022-03-21:
configuration blocks the work , cc1101 (rx) -GPIO 0  switches to the programming mode .
cc1101(TX)-GPIO2 interferes with work UART
please compile for others GPIO
I am a student, the project for arduino can be viewed with uncommented lines.

DigiH, 2022-03-21:
When compiling your builds in the Arduino IDE or platformio you can define your own PIN DEFINITIONS as you require in

https://github.com/1technophile/OpenMQTTGateway/blob/development/main/config_RF.h

Hope this helps

Monya2016, 2022-03-21:
Thanks ! # delete in what lines for cc1101?

DigiH, 2022-03-21:
Sorry, I don't quite understand that question, but maybe if you have a look at the CC1101 related documentation regarding the pinouts, as well as the required **ZradioCC1101** it becomes clearer

https://docs.openmqttgateway.com/setitup/rf.html#cc1101-pinout

In case you're using platformio you might also want to have a look at the predefined environment in platformio.ini

[env:nodemcuv2-rf-cc1101](https://github.com/1technophile/OpenMQTTGateway/blob/development/platformio.ini#L1019)

Monya2016, 2022-03-21:
I use for ESP8266 + CC1101 
https://docs.openmqttgateway.com/upload/web-install.html

Monya2016, 2022-03-21:
https://www.youtube.com/watch?v=QKiz63OB6Xo&t=286s
is this the correct compilation?

Monya2016, 2022-03-21:
in my case, connecting the receiver through a 5k resistor helped

closed 2023-10-16 by github-actions[bot]