1technophile/OpenMQTTGateway #1192: Renaming Miflora device (HHCCJCY01HHCC) in HA doesn't rename the entity IDs, even if specified

mihsu81, 2022-03-26:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
Renaming Miflora device (HHCCJCY01HHCC) in HA doesn't rename the entity IDs, even if specified.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit Miflora device and change the name
[image]
2. Click Rename when asked to rename entity IDs
[image]
3. Entity IDs keep the original name
[image]

**Expected behavior**
Entity IDs should change from the generic name to the specified name of the device.
[image]

**Screenshots**
Added under the previous steps. I think it helps better explaining the issue.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.11)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)
* For IR and RF clarify if you tested with the basic examples given with these libraries

**Additional context**
Add any other context about the problem here.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2022-03-26:
I think this is due to the fact that your device is created through Mqtt auto discovery.
If you want to have a particular entity id I suggest to define your sensor with yaml following the examples below:
https://docs.openmqttgateway.com/integrate/home_assistant.html#mijia-thermometer-ble

mihsu81, 2022-03-26:
Hi @1technophile,
Probably yes, but on the other hand, the entities of a LYWSD03MMC with PVVX firmware get renamed fine when i rename the device.
[image]
[image]
[image]

TBail, 2022-04-24:
Wouldn't it be an option to use the mac adress of an sensor and not the model name? I have 15 Miflora sensors and now a lot of entities starting with hhc... and if we add the mac to the device name as well it should really easy to identifiy things

mihsu81, 2022-04-28:
> I think this is due to the fact that your device is created through Mqtt auto discovery. If you want to have a particular entity id I suggest to define your sensor with yaml following the examples below: https://docs.openmqttgateway.com/integrate/home_assistant.html#mijia-thermometer-ble

Yeah, but auto discovery is more convenient, and renaming works fine for model LYWSD03MMC_PVVX. Just for HHCCJCY01HHCC doesn't work properly.

Could this really be caused by HA and not the implementation of auto discovery in OpenMQTTGateway?

closed 2022-06-12 by mihsu81