1technophile/OpenMQTTGateway #1203: MQTT OTA Update not working because of certificate issues

eloo, 2022-04-17:
**Describe the bug**
Trying to use the MQTT OTA update i see a cert error for downloading the file.
I have tried the script it self or manually sending the message. Also i have updated the PEM but still the cert can not be validated.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the ota_command_cert.sh from the website
2. Set a new version like "0.9.11" 
3. Run the script
4. Check the logs an you will see a certificate error

**Expected behavior**
Running the MQTT OTA update is working as described in the docs

**Screenshots**
-
**Environment (please complete the following information):**
* OpenMQTTGateway version: 0.9.10 prebuild from github release page
**Additional context**

Logs:

```
N: [ MQTT->OMG ]: {"version":"test","password":"OTAPASSWORD","url":"https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9.11/esp32dev-ble-firmware.bin","server_cert":"-----BEGIN CERTIFICATE-----\nMIIDxTCCAq2gAwIBAgIQAqxcJmoLQJuPC3nyrkYldzANBgkqhkiG9w0BAQUFADBs\nMQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3\nd3cuZGlnaWNlcnQuY29tMSswKQYDVQQDEyJEaWdpQ2VydCBIaWdoIEFzc3VyYW5j\nZSBFViBSb290IENBMB4XDTA2MTExMDAwMDAwMFoXDTMxMTExMDAwMDAwMFowbDEL\nMAkGA1UEBhMCVVMxFTATBgNVBAoTDERpZ2lDZXJ0IEluYzEZMBcGA1UECxMQd3d3\nLmRpZ2ljZXJ0LmNvbTErMCkGA1UEAxMiRGlnaUNlcnQgSGlnaCBBc3N1cmFuY2Ug\nRVYgUm9vdCBDQTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAMbM5XPm\n+9S75S0tMqbf5YE/yc0lSbZxKsPVlDRnogocsF9ppkCxxLeyj9CYpKlBWTrT3JTW\nPNt0OKRKzE0lgvdKpVMSOO7zSW1xkX5jtqumX8OkhPhPYlG++MXs2ziS4wblCJEM\nxChBVfvLWokVfnHoNb9Ncgk9vjo4UFt3MRuNs8ckRZqnrG0AFFoEt7oT61EKmEFB\nIk5lYYeBQVCmeVyJ3hlKV9Uu5l0cUyx+mM0aBhakaHPQNAQTXKFx01p8VdteZOE3\nhzBWBOURtCmAEvF5OYiiAhF8J2a3iLd48soKqDirCmTCv2ZdlYTBoSUeh10aUAsg\nEsxBu24LUTi4S8sCAwEAAaNjMGEwDgYDVR0PAQH/BAQDAgGGMA8GA1UdEwEB/wQF\nMAMBAf8wHQYDVR0OBBYEFLE+w2kD+L9HAdSYJhoIAu9jZCvDMB8GA1UdIwQYMBaA\nFLE+w2kD+L9HAdSYJhoIAu9jZCvDMA0GCSqGSIb3DQEBBQUAA4IBAQAcGgaX3Nec\nnzyIZgYIVyHbIUf4KmeqvxgydkAQV8GK83rZEWWONfqe/EW1ntlMMUu4kehDLI6z\neM7b41N5cdblIZQB2lWHmiRk9opmzN6cN82oNLFpmyPInngiK3BD41VHMWEZ71jF\nhS9OMPagMRYjyOfiZRYzy78aG6A9+MpeizGLYAiJLQwGXFK3xPkKmNEVX58Svnw2\nYzi9RKR/5CYrCsSXaQ3pjOLAEFe4yHYSkVXySGnYvCoCWw9E1CAx2/S6cCZdkGCe\nvEsXCS+0yx5DaMkHJ8HSXPfqIbloEpw8nL+e/IBcm2PN7EeqJSdnoDfzAIJ9VNep\n+OkuE6N36B9K\n-----END CERTIFICATE-----\n"}
W: Starting firmware update
N: Stop BLE processing
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -9984
[E][zzHTTPUpdate.cpp:212] handleUpdate(): HTTP error: connection refused
```

mihsu81, 2022-05-06:
I can confirm it worked for me.

eloo, 2022-05-06:
@nonflammable thanks.. yep this has worked..
but can you give a hint where you get this cert from?

@mihsu81 but can you explain where in the docs i get the cert?

mihsu81, 2022-05-07:
@eloo it's the certificate of DigiCert Global Root CA, the root certification authority which issued the certificate of the subordinate certification authority which in turn issued the current certificate of github.com. 
There are a lot of articles online which how you how to get the certificate in Base64 format.
I does make sense to change the instructions to use the current DigiCert Global Root CA certificate.

PR #1213 (2022-05-13): Update certificate in the examples for OTA

## Description:
update certificate to the root authority
fix #1203 

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [X] I accept the [DCO](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/participate/development.md#developer-certificate-of-origin).
closed 2022-05-13 by 1technophile