1technophile/OpenMQTTGateway #1205: RFBridge constant reboot/exception in 0.9.11

sergeolkhovik, 2022-04-24:
I have Sonoff RF bridge (with hw modification) and it ran 0.9.3 (self-compiled with some settings, including MQTT server configured as DNS name, not an IP), so I decided to reflash with some fresh version and I found that version 0.9.11 constantly throws an exception and reboots.
Here's some MQTT log:
```
кві 24 23:57:09 homeassistant/N6/LWT offline
кві 24 23:57:09 homeassistant/N6/LWT online
кві 24 23:57:09 homeassistant/N6/version 0.9.11
кві 24 23:57:09 homeassistant/binary_sensor/600194CC3BBC-connectivity/config {"stat_t":"homeassistant/N6/LWT","name":"SYS: Connectivity","uniq_id":"600194CC3BBC-connectivity","pl_on":"online","pl_off":"offline","pl_avail":"online","pl_not_avail":"offline","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/sensor/600194CC3BBC-uptime/config {"stat_t":"homeassistant/N6/SYStoMQTT","unit_of_meas":"s","name":"SYS: Uptime","uniq_id":"600194CC3BBC-uptime","val_tpl":"{{ value_json.uptime }}","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/sensor/600194CC3BBC-freemem/config {"stat_t":"homeassistant/N6/SYStoMQTT","unit_of_meas":"B","name":"SYS: Free memory","uniq_id":"600194CC3BBC-freemem","val_tpl":"{{ value_json.freemem }}","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/sensor/600194CC3BBC-ip/config {"stat_t":"homeassistant/N6/SYStoMQTT","name":"SYS: IP","uniq_id":"600194CC3BBC-ip","val_tpl":"{{ value_json.ip }}","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/sensor/600194CC3BBC-rssi/config {"stat_t":"homeassistant/N6/SYStoMQTT","unit_of_meas":"dB","name":"SYS: Rssi","uniq_id":"600194CC3BBC-rssi","val_tpl":"{{ value_json.rssi }}","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/switch/600194CC3BBC-restart/config {"stat_t":"homeassistant/N6/LWT","name":"SYS: Restart gateway","uniq_id":"600194CC3BBC-restart","pl_on":"{\"cmd\":\"restart\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"homeassistant/N6/commands/MQTTtoSYS/config","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/switch/600194CC3BBC-erase/config {"stat_t":"homeassistant/N6/LWT","name":"SYS: Erase credentials","uniq_id":"600194CC3BBC-erase","pl_on":"{\"cmd\":\"erase\"}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"homeassistant/N6/commands/MQTTtoSYS/config","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/switch/600194CC3BBC-discovery/config {"name":"SYS: Auto discovery","uniq_id":"600194CC3BBC-discovery","retain":true,"pl_on":"{\"discovery\":true}","pl_off":"{\"discovery\":false}","pl_avail":"online","pl_not_avail":"offline","cmd_t":"homeassistant/N6/commands/MQTTtoSYS/config","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/sensor/600194CC3BBC-gatewaySRFB/config {"stat_t":"homeassistant/N6/SRFBtoMQTT","name":"gatewaySRFB","uniq_id":"600194CC3BBC-gatewaySRFB","val_tpl":"{{ value_json.value | is_defined }}","device":{"identifiers":["600194CC3BBC"],"name":"N6","model":"[\"SRFB\"]","manufacturer":"OMG_community","sw_version":"0.9.11"}}
кві 24 23:57:09 homeassistant/N6/SYStoMQTT {"uptime":14,"version":"0.9.11","freemem":38976,"mqttport":"1883","mqttsecure":false,"rssi":-90,"SSID":"o-nest","ip":"xx","mac":"60:01:94:CC:3B:BC","modules":["SRFB"]}
кві 24 23:57:18 homeassistant/N6/LWT offline
кві 24 23:57:18 homeassistant/N6/LWT online
```
And here's the exception info:
```
Exception (9):
epc1=0x4020367b epc2=0x00000000 epc3=0x00000000 excvaddr=0x008c6eff depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9e0 end: 3fffffc0 offset: 0190
3ffffb70:  4020f120 00000000 3ffffb90 40203677  
...
3fffffb0:  feefeffe feefeffe 3ffe87c4 40101465  
<<<stack<<<
```
I did decoding and here's the info:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x4020367b
EXCVADDR: 0x008c6eff
```
That is rfbridge-firmware.bin, and I even tried to recompile 0.9.11 with my settings:
```
[env:rfbridge]
platform = ${com.esp8266_platform}
board = esp8285
lib_deps =
  ${com-esp.lib_deps}
#  ${libraries.wifimanager8266}
  ${libraries.esp8266_mdns}
#  ${libraries.rc-switch}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
;  '-DZgatewaySRFB="SRFB"'
  '-DOMG_VERSION="0.9.11"'
  '-DRF_RECEIVER_GPIO=4'
  '-DRF_EMITTER_GPIO=5'
  '-DLED_INFO=13'
  '-DLED_INFO_ON=0'
  '-DGateway_Name="N6"'
  '-DBase_Topic="homeassistant/"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="xx"'
  '-Dwifi_password="xx"'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_SERVER="xx"'
  '-UMQTTsetMQTT'
```
The problem is still here.
OK, I decided to flash precompiled rfbridge 0.9.3 and all is fine, 0.9.4 runs OK as well. 0.9.5 fails, after some short time it stops responding, here's the log
```
*WM: [1] Connecting to new AP: xx
*WM: [3] Using Password: xx
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] xx
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
{"mqtt_server":"xx","mqtt_port":"1883","mqtt_user":"","mqtt_pass":"","mqtt_topic":"homeassistant/","gateway_name":"OpenMQTTGateway_SRFB"}N: Setup OpenMQTTGateway end
       W: MQTT connection...
                            N: Connected to broker
                                                  N: Subject: /SYStoMQTT
                                                                        N: Received json : {"uptime":56,"version":"v0.9.5","freemem":36672,"rssi":-59,"SSID":"o-nest","ip":"xx","mac":"60:01:94:CC:3B:BC","wifiprt":0,"modules":"SRFBHADiscovery"}

��U
N: Received json : {}

��U

��U

��U

��U
```
And then AP with very strange name (like �) appears on my phone (looks like it writes something wrong into flash storage).

DigiH, 2022-04-24:
Hi, I think it could be

#  ${libraries.wifimanager8266}

Being commented out. 

Introduced in 0.9.9 under BREAKING. 

Hope this helps

sergeolkhovik, 2022-04-25:
@DigiH thanks for the note, but that might work only for self-build image. I have problems only with RF module, when I flash [sonoff-rfbridge-direct-firmware.bin](https://github.com/1technophile/OpenMQTTGateway/releases/download/v0.9.11/sonoff-rfbridge-direct-firmware.bin) that includes Pilight - all is fine, no exception and reboot. But I need RF due to Digoo PIR sensor.
I think  I'm OK even with 0.9.4 - all my sensors in that location work OK, the only thing that stopped working - RF relay (don't know why, may be version, may be anything else), but I replaced it with Sonoff Basic + ESPHome and all is fine now (even better, now I know the state of the relay).

DigiH, 2022-04-25:
@sergeolkhovik Sorry, I thought you made a self-built image for that with the above environment details ;) 

```
[env:rfbridge]
platform = ${com.esp8266_platform}
board = esp8285
lib_deps =
  ${com-esp.lib_deps}
#  ${libraries.wifimanager8266}
  ${libraries.esp8266_mdns}
#  ${libraries.rc-switch}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayRF="RF"'
;  '-DZgatewaySRFB="SRFB"'
  '-DOMG_VERSION="0.9.11"'
  '-DRF_RECEIVER_GPIO=4'
  '-DRF_EMITTER_GPIO=5'
  '-DLED_INFO=13'
  '-DLED_INFO_ON=0'
  '-DGateway_Name="N6"'
  '-DBase_Topic="homeassistant/"'
  '-DESPWifiManualSetup=true'
  '-Dwifi_ssid="xx"'
  '-Dwifi_password="xx"'
  '-DMQTT_USER=""'
  '-DMQTT_PASS=""'
  '-DMQTT_SERVER="xx"'
  '-UMQTTsetMQTT'
```

Did you also try basing it on the **sonoff-rfbridge-direct** environment and making your changes/additions there, instead of using the **rfbridge** environment? Especially with the platform definition differences.

All the best

closed 2023-10-16 by github-actions[bot]