1technophile/OpenMQTTGateway #1206: Support for hooks / intended side effects

mpiccolino-tealblue, 2022-04-28:
I am working on a "soil moisture over LoRa" project where MQTT connectivity with a broker should be optional, not required.
The idea is to have a low-cost solution for small veg garden projects, which can then be expanded if a local Controller and/or an internet connection are available.

Hence, I would like the gateway (a Heltec ESP32 v2) to perform a REST call directly to a wifi-enabled sprinkler every time a LoRa message is received, while sending it also to the MQTT broker.

Is this already supported by OpenMQTTGateway without messing too much with the core code?

I'm looking into other low-code solutions, like homie-esp8266, which have hooks of this type, but are tied to the ESP architecture.

Thanks for any feedback!

1technophile, 2022-04-28:
> Is this already supported by OpenMQTTGateway without messing too much with the core code?

Not supported out of the box. Mqtt connection is required for OMG to run. You would have to modify the code to do what you want.

mpiccolino-tealblue, 2022-04-29:
Thanks for the feedback!