1technophile/OpenMQTTGateway #1212: LYWSD03MMC_PVVX are not auto detected

benzok, 2022-05-05:
**Describe the bug**
First I want to say; This is my first Bug Report, so be gentle ;)

I have used two LYWSD03MMC_PVVX connected via OMG to HA for ~6 months all working fine, I just connected another two but unfortunately they did not appear in HA.

The current values för Temp, Humidity and Battery where published on `BTtoMQTT`, but nothing under `homeassistant/sensor`

I built with TRACE and looking through the log I could see that the sensors where classified as UNKNOWN_MODEL
```
T: Device mac A4:C1:38:22:E1:B9
T: Device UNKNOWN_MODEL A4:C1:38:22:E1:B9
```
After a while (seconds) they are detected 

```
N: Device detected: A4:C1:38:22:E1:B9
T: getDeviceByMac A4:C1:38:22:E1:B9
T: Get services data number: 1
T: Converted service data (15) to b9e12238c1a488075710810b52af0f
T: Service data: b9e12238c1a488075710810b52af0f
T: Service data UUID: 0x181a
T: Decoder found device: LYWSD03MMC_PVVX
T: getDeviceByMac A4:C1:38:22:E1:B9
T: update A4:C1:38:22:E1:B9
T: Removing servicedatauuid
T: Removing servicedata
T: Device mac A4:C1:38:22:E1:B9
T: Looking for Model_id: 27
```
Unfortunately there is no auto discovery message.

Looking into the code, I guess that the device classified as UNKNOWN_MODEL has `device->isDisc` set to true, so I did the following change

```
diff --git a/main/ZgatewayBT.ino b/main/ZgatewayBT.ino
index 2035f05..839f09f 100644
--- a/main/ZgatewayBT.ino
+++ b/main/ZgatewayBT.ino
@@ -267,6 +267,10 @@ void createOrUpdateDevice(const char* mac, uint8_t flags, int model, int mac_typ
     }

     if (model != TheengsDecoder::BLE_ID_NUM::UNKNOWN_MODEL) {
+      if (device->sensorModel_id == TheengsDecoder::BLE_ID_NUM::UNKNOWN_MODEL) {
+        device->isDisc = false;
+        newDevices++;
+      }
       device->sensorModel_id = model;
     }
```

If this is the right approach or if the error lies within OMG or in the pvvx firmware I don't know. Nevertheless with this change the Mijia sensors are auto detected.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash an LYWSD03MMC with pvvx v3.7d
2. Add mac-address to white-list
3. Wait for a message on `homeassistant/sensor` (to no avail)

**Expected behavior**
An auto detect message should appear on `homeassistant/sensor`

**Environment **
* OpenMQTTGateway version used V0.9.12 flashed on an esp32-lolin32lite-ble
* Xiaomi Thermometer flashed with pvvx v3.7d

1technophile, 2022-05-08:
Hello,

Thanks for this detailed report, perfect for a first one :-)

I'm suspecting that some message of the PVVX firmware are not recognized, leading to having this sensor categorized as unknown. Would it be possible to connect to your broker with mqtt explorer and record the history of messages during 5min of the pvvx firmware.
You can post it there after.

benzok, 2022-05-10:
Here is what I think you asked about, but I don't really understand how this could help.
I switched off the OMG, started the subscription and then powered on the OMG
There are no messages on any other topic

```
 mosquitto_sub -h home.local -t 'home/OpenMQTTGateway1/BTtoMQTT/A4C13822E1B9' | jq .
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -73,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.31,
  "tempf": 70.358,
  "hum": 39.99,
  "batt": 83,
  "volt": 2.947
}
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -71,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.35,
  "tempf": 70.43,
  "hum": 39.99,
  "batt": 83,
  "volt": 2.947
}
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -70,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.33,
  "tempf": 70.394,
  "hum": 39.95,
  "batt": 82,
  "volt": 2.946
}
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -71,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.31,
  "tempf": 70.358,
  "hum": 39.98,
  "batt": 83,
  "volt": 2.947
}
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -76,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.31,
  "tempf": 70.358,
  "hum": 39.97,
  "batt": 82,
  "volt": 2.946
}
{
  "id": "A4:C1:38:22:E1:B9",
  "mac_type": 0,
  "name": "ATC_22E1B9",
  "rssi": -71,
  "brand": "Xiaomi",
  "model": "LYWSD03MMC",
  "model_id": "LYWSD03MMC_PVVX",
  "tempc": 21.31,
  "tempf": 70.358,
  "hum": 39.96,
  "batt": 83,
  "volt": 2.948
}

```

1technophile, 2023-04-10:
@benzok could you try with latest version please

1technophile, 2023-06-12:
Closing feel free to reopen if any question

closed 2023-06-12 by 1technophile