1technophile/OpenMQTTGateway #1215: Removal of Bluetooth device type TPMS

mihsu81, 2022-05-14:
**Is your feature request related to a problem? Please describe.**
Bluetooth TMPS devices keep appearing when HA Auto Discovery is enabled.

**Describe the solution you'd like**
Removal of Bluetooth device type TPMS, as already done in v0.9.12 for GAEN, MS_CDP and iBeacon.

**Describe alternatives you've considered**
.

**Additional context**
[image]
[image]

1technophile, 2022-05-14:
I'm suspecting a bug on the decoder, you should not have [TPMS](https://github.com/theengs/decoder/blob/development/src/devices/TPMS_json.h) entities created if you don't have a BLE TPMS.

Are those devices created with v0.9.12 or where there prior to it?

mihsu81, 2022-05-15:
The entities are created with v0.9.12. They appeared a couple of days ago, while the update to v0.9.12 was performed approximately a week ago.

1technophile, 2022-05-19:
@mihsu81 Could you try this new version of the decoder and check if you still see TPMS ?
https://github.com/theengs/decoder

mihsu81, 2022-07-02:
Sorry for the delay. Completely forgot about it. I've upgraded the gateways to v0.9.13. Let's see if the TPMS devices reappear.

1technophile, 2023-02-05:
Should be good with the new versions

closed 2023-02-05 by 1technophile