1technophile/OpenMQTTGateway #1219: BLE Mesh support

l1bbcsg, 2022-05-24:
As far as I understand OpenMQTTGateway does not BLE Mesh currently?

Are there any plans to support it?

BLE mesh is a protocol distinct from BLE. In a nutshell, it's a mesh network competing with Zigbee.
Hardware support is universal though, both popular chips and SDKs from Espressif and Nordic support Mesh out of the box.

Running as BLE Mesh Proxy node would a perfect use case scenario for a MQTT broker. Currently there are no easy ways to integrate BLE Mesh into home automation systems.

h2zero, 2022-05-24:
While this would be a nice addition it would be a lot of work and probably not worth it as there are very few devices (that I know of) with mesh capability.

What devices do you have that you'd like to communicate with?

1technophile, 2023-02-05:
Feel free to answer and reopen if needed

closed 2023-02-05 by 1technophile