1technophile/OpenMQTTGateway #1221: ssl wrong version number

fariouche, 2022-06-01:
**Describe the bug**
I'm using a secured mqtt broker (mosquitto) and it is set up to only accept tls (default behavior)
I'm getting an error "ssl wrong version number" which indicates that the esp32 is not using tls (most likely from what I've read)

I know that the CA is valid, as it works with the mosquitto_sub command line

My serial traces are weird, (carriage return) but here are the traces (no error) and it will continue to display n24. then n34. etc
I don't know yet were to look at in the code to investigate.

*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.30.134
N: BLE scans interval: 55555
                            N: BLE scans number before connect: 10
                                                                  N: Publishing only BLE sensors: false
                                                                                                       N: minrssi: 100
                                                                                                                      N0
                                                                                                                       ]
                                                                                                                       *
                                                                                                                       .
                                                                                                                       n
                                                                                                                       1
                                                                                                                       4
                                                                                                                       .

**To Reproduce**
Steps to reproduce the behavior:
1- Setup an mqtt broker to use ssl (set keyfile and certfile and cafile and listener is set to 8883. All other fields are commented. No username etc)
2- Setup User_config.h and set certificate, MQTT_SERVER and MQTT_PORT
3- when setuping the OMG using the portal, I've made sure I've set secure and removed user and password.
4- connect to the broker
5- see the error from mosquitto: "OpenSSL Error[0]: error:1408F10B:SSL routines:ssl3_get_record:wrong version number"

I don't know yet if I've done something wrong (not impossible even after triple checking!)

**Expected behavior**
OMG should connect using the provided ca file and settings

**Environment (please complete the following information):**
* OpenMQTTGateway version used : development

fariouche, 2022-06-02:
I've got it working by hardcoding the certificate in User_config.h.
Pasting it in the portal seems to corrupt it (mine is 4096 Bytes)

1technophile, 2022-06-07:
@fariouche did you tested if it works?

fariouche, 2022-06-09:
Ok, good news, it works with 4096 :)

g0lgs, 2023-03-16:
V1.4.0 has something that looks odd in line 1277 of main.ino

`WiFiManagerParameter custom_mqtt_cert("cert", "<br/>mqtt broker cert", mqtt_cert.c_str(), 2048);`

As my cert is also 4096 bytes I changed this to 

`WiFiManagerParameter custom_mqtt_cert("cert", "mqtt broker cert", mqtt_cert.c_str(), 4096);`

my device is now connecting fine

1technophile, 2023-03-16:
Thanks for the info, feel free to submit a PR with the change

g0lgs, 2023-03-16:
> Thanks for the info, feel free to submit a PR with the change

I am not familar with creating PR's (never done it).